pixelmatix/SmartMatrix #21: FeatureDemo.ino:6:31: fatal error: SmartMatrix_16x32.h: No such file or directory

ners11, 2015-07-09:
Please help me.
[image]

embedded-creations, 2015-07-09:
It doesn't look like you have the SmartMatrix_16x32 library installed properly.  Do you see SmartMatrix_16x32 in your Arduino File->Example or Sketch->Include Library menus?

Installation instructions are here:
http://docs.pixelmatix.com/SmartMatrix/#software-and-teensy-setup

ners11, 2015-07-09:
Yes, I do.

jasoncoon, 2015-07-09:
Please enable verbose compilation output in Arduino, and then copy and paste the entire output to pastebin: http://pastebin.com

File->Preferences->Show verbose output during: Tick box for compilation

jasoncoon, 2015-07-09:
Also, make sure you've restarted the Arduino IDE since you installed the library.  And I'd recommend updating to the latest version of the Arduino IDE (1.6.5): https://www.arduino.cc/en/Main/Software

ners11, 2015-07-09:
http://pastebin.com/TqsUBAy7

ners11, 2015-07-09:
All works, thanks for the help !

closed 2015-07-15 by embedded-creations