pixelmatix/SmartMatrix #23: Support for smaller color spaces

embedded-creations, 2015-07-15:
Part of @mrwastl's #17 
https://github.com/mrwastl/SmartMatrix/commit/a85836fe27a4f080a42242d0d7983c05775af62a
https://github.com/mrwastl/SmartMatrix/commit/a280bf6e7e311dfa705426e974a2db2c8a5f7b3b

embedded-creations, 2015-09-08:
@mrwastl we have a place to discuss smaller color spaces here.

you should be able to replace this code in SMLayerBackground<RGB, optionFlags>::fillRefreshRow
```
            refreshRow[i] = rgb48(backgroundColorCorrectionLUT[currentPixel.red],
                backgroundColorCorrectionLUT[currentPixel.green],
                backgroundColorCorrectionLUT[currentPixel.blue]);
```
with 
```
            colorCorrection(currentPixel, refreshRow[i]);
```
Add a new colorCorrection function for your smaller color spaces, and it should work.  All the rest of the changes should be in MatrixCommon.h if I'm not mistaken.

mrwastl, 2015-10-05:
@embedded-creations  just that i get it right:

* indexed_layer: 1 bit / pixel, double buffered, storage_depth has no influence
* scrolling layer: 1 bit / pixel, single buffered, storage_depth has no influence
* background layer: storage_depth bits / pixel, double buffered
* allocate buffers: buffer space for 'mixing' layers, colour correction, pwm_depth is responsible for level of detail when generating colours

* rgb24/rgb48 is relevant for background layer but not so for pwm generating - so if i add colour spaces i only need to focus on background layer + converting colour information from / to  the different layers used (if for example pwm_depth is 24 and  storage_depth is 8, colour information that is requested while calculating the pwm buffers needs to be converted by rgb8 to rgb24).

i've already played around with rgb8 and rgb16, but there must be an error in reasoning or so somewhere ...  (usually in a place where i don't look ;-)

embedded-creations, 2015-10-07:
@mrwastl Yes, that looks correct.  Have you tested out your rgb8/16 conversion functions?  Try isolating some code, maybe you could change working storage_depth=24 code to convert rgb24 to rgb16 and back to rgb24 within the fillRefreshRow() function to see if it's the conversion functions or something else that may not be working properly.

mrwastl, 2015-10-10:
@embedded-creations 
do you have an idea, why the ``operator=``-methods found in your library aren't working with implicit casts (i don't have that much experience with operator-overloading with structs):

```
  rgb8 colred8 = {255, 0, 0};
  rgb24 colred = colred8;
```

the 2nd assignment isn't working. i get this error-message:
``error: conversion from 'rgb8' to non-scalar type 'rgb24' requested``

but if i add an additional cast-constructor, above example is working:

```
typedef struct rgb24 {
...
    rgb24& operator=(const rgb8& col);
    rgb24& operator=(const rgb16& col);
    rgb24& operator=(const rgb24& col);
    rgb24& operator=(const rgb48& col);

    // cast-constructors added
    rgb24( const rgb8& col );
    rgb24( const rgb16& col );
    rgb24( const rgb24& col );
    rgb24( const rgb48& col );
...
}
```

``rgb24& operator=(const rgb8& col);`` and ``rgb24( const rgb8& col );`` contain exactly the same code (except an additional ``return *this;`` in ``operator=``).

it would be logical if the ``operator=`` overloading methods (defined for all supported colour-types!) would just do that?

maybe you know a better way how to solve this. or how to avoid code redundancy.

embedded-creations, 2015-10-13:
@mrwastl Sorry I have no idea.  This is beyond my C++ experience, we need an expert.

@gregfriedland do you have any idea why the implicitly defined assignment operators for `rgb*` types aren't working?  For rgb24, I needed to create an assignment operator or drawing to the last pixel of the background buffer would crash the Teensy:
https://github.com/pixelmatix/SmartMatrix/blob/sm3.0/src/MatrixCommon.h#L64

gregfriedland, 2015-10-13:
@pixelmatix: it's hard to tell without looking at the call line but my guess is that the other assignment operator was getting triggered somehow and it was reading an rgb24 as if it were an rgb48.

Sent from my phone. Please forgive any terseness.

mrwastl, 2015-10-15:
i've pushed a new branch: [``sm3.0-colourspaces``](https://github.com/mrwastl/SmartMatrix/tree/sm3.0-colourspaces).

this branch adds support for:
* ``rgb8`` (RGB332)
* ``rgb16`` (RGB565)

colour space can be changed for layers (``COLOR_DEPTH`` aka ``storage_depth``), but not for ``pwm_depth`` (aka ``kRefreshDepth``). 

tested and working:
* ``rgb8`` and ``rgb16`` together with ``kRefreshDepth = 24``
* colour correction works with ``kRefreshDepth = 24`` (but crashes with ``kRefreshDepth = 36`` and ``=48`` at the moment (problem identified (out of bounds error when accessing the colour correction LUT), but not fixed yet)
* backgroundLayer

not yet working or not implemented yet:
* ``indexedLayer`` and ``scrollingLayer`` (next thing on my list)
* colour correction for ``kRefreshDepth > 24``

drawbacks:
* slower than ``rgb24`` at the moment (because of colour space conversions?)
  test video: 44 fps with ``COLOR_DEPTH=24``, but ~ 33 fps with ``=8`` or ``=16``

differences to original sm3.0-branch:
https://github.com/mrwastl/SmartMatrix/compare/sm3.0...mrwastl:sm3.0-colourspaces

embedded-creations, 2015-10-16:
Great!  Do you think there's much of a reason to add support to `indexedLayer` and `scrollingLayer`?  The memory saved in those layers by using smaller colors is only going to be a couple bytes.  Maybe there's a good reason for consistency.

> slower than rgb24 at the moment

I wonder if there are any tricks people have come up with (maybe using assembly instead of C code) in the past to quickly unpack the individual color channel bits in rgb8 or rgb16

mrwastl, 2015-10-17:
new push, see corrected / extended posting above.

there aren't any memory savings for indexedLayer nor scrollingLayer (1 bit / pixel can hardly be improved :), but some small conversion fixes had to be done.

speed: well, a little drawback, but still usable. maybe there're some unnecessary implicit colour space casts.

normalised colour initialisation: i've added constructors that allow setting colour space independent colour values (range [0.0 - 1.0] for each colour channel).
problem: c/c++ cannot differentiate between constructor({integer, integer, integer}) and constructor({float, float, float}) (eg: RGB({ 255, 255, 255 })  and RGB({1.0, 1.0, 1.0})) - for c/c++ this is ambiguous ...
so i had to introduce a 4th dummy float that is not used ('t' as in 'transparency').

maybe there's a better solution but i couldn't find a better one ...

mrwastl, 2015-10-18:
again new commit (https://github.com/mrwastl/SmartMatrix/commit/fc4aa0786b8f599b904d534a69eba5ed073f4314):

* some small optimisation for ``rgb8`` and ``rgb16`` (direct access to colour information using a ``union``.
* cheap trick to extend the range when converting from rgb24 to rgb48.

ecurtz, 2015-10-26:
I've added the obvious and trivial
static void loadMatrixBuffers12(unsigned char currentRow, unsigned char freeRowBuffer);
in my local working copy, which would probably be a good fit here if you haven't done it already.

ecurtz, 2015-10-26:
Also, are you using a version of the old streaming demo for you testing? If so, using smarter Serial methods rather than reading a byte at a time makes a huge difference in the framerate.

  char* buffer = (char*)backgroundLayer.backBuffer();
  int bytesAvail = Serial.available();
  boolean swap = false;
  
  if ((bytesAvail > 1) && (dataPos > 0))
  {
    if (bytesAvail > dataExpected) bytesAvail = dataExpected;
    int count = Serial.readBytes(&buffer[dataPos], bytesAvail);
    dataPos += count;
    dataExpected -= count;
    if (dataExpected == 0) {
      swap = true;
    }
  }
  else if (bytesAvail) {
    char val = Serial.read();
    if ( val == 1 )
    {
      dataPos=0;
      dataExpected = kMatrixWidth * kMatrixHeight * 3;
    }
    else {
      buffer[dataPos++] = val;
      dataExpected--;
      if (dataExpected == 0) {
        swap = true;
      }
    }
  }

mrwastl, 2015-10-27:
@ecurtz:
``loadMatrixBuffers12``: nope. as written above my focus was on storage depth and not on pmw depth yet.  could you post your method so that i can test it in my branch?

serial transfer: my library (VSSDCP) that i use for transferring data is already sending chunks (256 bytes in one go for teensy).

ecurtz, 2015-10-27:
https://github.com/ecurtz/SmartMatrix/commit/30dd3c195a72a5f65becc70d0a1f1da396127ad0

mrwastl, 2015-10-28:
@ecurtz 
i've tested your new method, but there seems to be a problem with the correct usage/mapping of colours (eg. when testing FeatureDemo.ino: black background (instead of dark red), wrong colours used when drawing objects, ...)

i'll have a look at this somewhen later, but i'm currently working on a different project.

ecurtz, 2015-10-28:
Are you sure it isn't just the expected loss of color fidelity from reducing output from 8bit per channel to 4bit? The code is really basic.

mrwastl, 2020-09-08:
@embedded-creations
nice! now i no longer need to add this to my fork (or need to understand all teensy4 relevant changings :)

@ecurtz 
(sorry that my answer is 5 years late:) i (still) fool around with streaming video content to my displays (including rgb matrixes) and for those displays 4bit usually is a good compromise between video quality and throughput (and thus refreshrate).

embedded-creations, 2020-09-08:
@mrwastl I haven't fully integrated your changes yet, but plan to.  Some more details here:

https://community.pixelmatix.com/t/smartled-shield-for-teensy-4/740/12?u=louis

embedded-creations, 2020-09-08:
@mrwastl How are you streaming content?  Wireless?  USB?

mrwastl, 2020-09-08:
@embedded-creations 
i've implemented my own protocol (that I'm not very happy with): https://github.com/mrwastl/VSSDCP (github-version is very outdated; only serial/USB is working correctly at the moment, I will push my local changes that contain support for ESP32, different teensys a.s.o.). My library serdisplib contains support for this protocol and years ago I've made a patch for mplayer to add support for serdisplib (https://github.com/mrwastl/mplayer-vo_serdisp). With all these parts together I'm able to stream video content. 

But I think I'll have a look at TPM2 (suggested by  drbeefsupreme at the pixelmatrix-forum).

Wireless: I have tried to stream via wireless (ESP32, teensy using external modules) and ethernet (TCP/UDP), but the results have been very disappointing (lots of streaming errors).
And I already have an ethernet adapter for teeensy4 - so this will be one of my next experiments.

@embedded-creations 
(offtopic): i have tried to login at community.pixelmatrix.com using my github-account but I get a 'Sorry, there was an error authorizing your account. Please try again.' - Any idea what might cause this?

embedded-creations, 2020-09-14:
I committed a lot of changes to add two new layers with Adafruit_GFX support.  I posted details here:

https://community.pixelmatix.com/t/adafruit-gfx-compatible-layers-modern-font-support-coming-to-smartmatrix-library-4-0/768

@mrwastl Cool, I didn't know that serdisplib is your project!

I know WLED has some streaming support, but I haven't tried it yet.  This project looks promising too, specifically for using some lightweight compression on the video to increase the pixel count:

- https://twitter.com/spectrenoir06/status/1304564947062599681
- https://antoine.doussaud.org/leds_master
- https://github.com/spectrenoir06/OctoWifi-LEDs-Controller

I think the Login with GitHub feature is broken.  To be honest I didn't even know that was an option on the forum.  I recommend just signing up with an email.