pixelmatix/SmartMatrix #30: Font Improvements

embedded-creations, 2015-09-18:
The fonts in SmartMatrix could use some improvements.  I'm collecting a list of updates to make, please let me know if you have anything to add.

* Only compile in fonts that are used by the Sketch
* Consider switching from using bitmaps to more space efficient BDF, [like PJRC's graphics library does](https://forum.pjrc.com/threads/26305-Highly-optimized-ILI9341-(320x240-TFT-color-display)-library?p=79853&viewfull=1#post79853)
* Support >8-pixel wide fonts

ecurtz, 2015-09-18:
int getBitmapFontLocation(unsigned char letter, const bitmap_font *font) searches through an array mapping font entries -> character codes. If this was changed to an array of character codes -> font entries then it's simply an array lookup, and the array can be unit8 rather than the current uint16. If the font contains more than 255 characters an offset and second array could be used, but that seems unlikely. The array would need to be as large as the highest char code, but that's not an issue for any of the current fonts, and it could have an initial skip for < 32 and other tricks if space was an issue.

embedded-creations, 2015-09-18:
@ecurtz Makes sense.  I briefly looked at the font code and I think if we limit the fonts to a max character code of 255 (which they should be now), the array size can be changed to uint8_t, no major code changes needed.  There's a couple fonts like Font_apple6x10.c that have >255, but I think that's just because I was manually trimming down the table in the early days and forgot to delete codes >255.  Your method may have other efficiency improvements I haven't considered yet.

Before I spend any time improving the fonts I want to evaluate and try the code in the PJRC library as if there are substantial advantages I may end up replacing a lot of my code with Paul's code.

I have a number of other improvements that are ahead of Fonts, so I probably won't start working on this for a good while.

yoonghm, 2016-02-04:
I worked on Unicode before by converting my own Chinese fonts. Depending the languages (Chinese, Japanese, Korean, etc). I did not complete that portion as I was having problem to locate the font bitmap quickly as the array is big. The existing code search the code linearly, which is fine for 255 characters but not for 65535 characters.

I suggest this

- Use compile-time define to restrict the availability of fonts (uint8, uint16, uint32).
- The compile-time define also define the appropriate alias data type that is used in various font accessing function.

embedded-creations, 2016-02-08:
@yoonghm Thanks for the suggestions!

grene78, 2016-08-17:
Support for fonts > 8px wide is a huge one for me.  Have you done any work on that, have any suggestions?  I'd hate to pour time into that one if you're switching to BDF anytime soon.

embedded-creations, 2016-08-20:
@grene78 I haven't done anything work on this yet.

embedded-creations, 2017-03-30:
Some work on >8px fonts posted on the SmartMatrix Community: http://community.pixelmatix.com/t/improving-fonts/205

embedded-creations, 2020-09-14:
(I'm updating old issues mentioning fonts.)  SmartMatrix Library 4.0 will have much better support for custom fonts and larger fonts.  I'm posting more details here:

https://community.pixelmatix.com/t/adafruit-gfx-compatible-layers-modern-font-support-coming-to-smartmatrix-library-4-0/768

closed 2020-11-23 by embedded-creations