pixelmatix/SmartMatrix #31: colorCorrection(const RGB_IN& in, rgb24& out) bug

ecurtz, 2015-10-25:
colorCorrection(const RGB_IN& in, rgb24& out) is using the low bits of the table rather than the high:

fix would be changing the local to rgb48:
template <typename RGB_IN>
void colorCorrection(const RGB_IN& in, rgb24& out) {
    out = rgb48(lightPowerMap16bit[in.red],
                lightPowerMap16bit[in.green],
                lightPowerMap16bit[in.blue]);
}

or simply skipping the local:
template <typename RGB_IN>
void colorCorrection(const RGB_IN& in, rgb24& out) {
    out.red = lightPowerMap16bit[in.red] >> 8;
    out.green = lightPowerMap16bit[in.green] >> 8;
    out.blue = lightPowerMap16bit[in.blue] >> 8;
}

Sorry to not make a pull request, seemed like too much effort for one line.

embedded-creations, 2015-10-27:
Hi Eli, Thanks for this, I'll mark it down to fix for the next minor release.  No problem submitting it as an issue, not a pull request, plus you gave me two options on how to fix it which you couldn't do with a PR.

ecurtz, 2015-10-27:
I did finally make my own fork, so here's the commit anyway (I did the direct version rather than rgb48, because I'm old school, efficiency obsessed): https://github.com/pixelmatix/SmartMatrix/commit/9eedf6b4bb114a67f2013f0c1f92ea1541616f81

commit eb9eb35d (2020-11-24): Fix colorCorrection bug, thanks @ecurtz, fixes #31 and likely fixes #85
closed 2020-11-24 by embedded-creations