pixelmatix/SmartMatrix #32: all sample of Smart Matrix lib Modified by Jason Coon compile error

mkeyno, 2015-11-08:
I don't know why my arduino  (1.6.4) win installed FastLED lib along with SmartMatrix_32x32 & LEDMatrix  can compile well and issue the following compile error, any help really appreciated

D:\arduino-1.6.4-windows\arduino-1.6.4\libraries\FastLED_3.1.0/FastLED.h: In static member function 'static CLEDController& CFastLED::addLeds(CRGB*, int, int)':
D:\arduino-1.6.4-windows\arduino-1.6.4\libraries\FastLED_3.1.0/FastLED.h:326:32: error: 'CSmartMatrixController' does not name a type
    case SMART_MATRIX: { static CSmartMatrixController controller; return addLeds(&controller, data, nLedsOrOffset, nLedsIfOffset); }
                                ^
D:\arduino-1.6.4-windows\arduino-1.6.4\libraries\FastLED_3.1.0/FastLED.h:326:83: error: 'controller' was not declared in this scope
    case SMART_MATRIX: { static CSmartMatrixController controller; return addLeds(&controller, data, nLedsOrOffset, nLedsIfOffset); }

embedded-creations, 2015-11-09:
What sketch are you trying to compile?  I can't figure out what might be wrong from the information you provided.

There's a new release of the SmartMatrix Library, 3.0, and you can't use the FastLED Controller, but you can still write sketches that use FastLED.  I suggest upgrading to SmartMatrix 3.0 and looking at the FastLED_Functions example.

mkeyno, 2015-11-11:
thanks for reply buddy , I've just download  and add it  last SmartMatrix Library, 3.0,  and  FastLED but it seems the new library is not updated for new version of FastLED library and I thinks this issue related to this problem , 

have you check it by yourself your library  with with new FastLED library ?

embedded-creations, 2015-11-11:
Please tell me what sketch you are trying to compile so I can try to reproduce

embedded-creations, 2015-11-11:
Example sketch FastLED_Functions from SmartMatrix 3.0 compiles with FastLED 3.1

Make sure you're including "SmartMatrix3.h" and not "SmartMatrix.h" in your sketch.

closed 2020-11-23 by embedded-creations