pixelmatix/SmartMatrix #34: 1/4 scan mode

happycodefarm, 2016-01-09:
Would it be possible to implement a 1/4 scan mod for the 16x32 outdoor hub75 matrix ?

Regards,

HCF

embedded-creations, 2016-01-09:
Yes, I have a sample of an indoor 1/4 scan 16x32, and it's on my list though low priority.

If you're comfortable modifying the library yourself, I think I made it pretty easy to do so, just look for where DIV16 and DIV8 options are defined and used and add a DIV4.  Post here or on our forum if you need help: community.pixelmatix.com

embedded-creations, 2016-01-09:
Do you have a link to purchase the outdoor hub75 matrix you have?  If it's cheap I might pick one up as a reference

embedded-creations, 2016-01-09:
Also, I checked our your Psychedelic Pixels project, you might be interested in the port of Fadecandy to SmartMatrix, if you haven't seen it already - you can stream visuals from Processing to a matrix.  I haven't advertised it much yet as the documentation isn't done:
https://github.com/pixelmatix/SmartMatrix_Fadecandy

embedded-creations, 2020-11-23:
This is a really old issue, but just in case someone's still looking at it, search for MultiRowRefreshMapping in community.pixelmatix.com

closed 2020-11-23 by embedded-creations