pixelmatix/SmartMatrix #37: FeatureDemo won't compile

nvisible, 2016-04-18:
I am using a brand new SmartMatrix3 shield and the FeatureDemo won't compile in the Arduino IDE. I can run other sketches on the Teensy 3.2 board but none of the SmartMatrix3 library examples are compiling. I've tried different Teensy boards in the IDE as well as clock speeds. Nothing seems to work. I'm building a dashboard for an electric motorcycle build using a 64x32 P3 array from Adafruit.

Error messages attached.
[TeensyError.txt](https://github.com/pixelmatix/SmartMatrix/files/224758/TeensyError.txt)

embedded-creations, 2016-04-18:
Quick answer is it probably has to do with Arduino 1.6.8, Teensyduino 1.28.  I haven't tried past 1.6.7, 1.27.  Downloading now.

embedded-creations, 2016-04-18:
Confirmed, I see this with Arduino 1.6.8, Teensyduino 1.28.  I recently saw this error pattern when compiling SmartMatrix for the Particle Photon (an APA102 LED strip port of SmartMatrix, not the whole SmartMatrix Library), and fixed it, but can't remember how I fixed it.  I'll investigate when I have time but for now, please download 1.6.7, 1.27 to get unstuck.

embedded-creations, 2016-04-18:
I just pushed a fix.  I'll package this into a release when I have time, but you can either download the master branch, or just navigate into your libraries directory and add `#include <stdlib.h>` to the top of src/Layer_Background_Impl.h

embedded-creations, 2016-04-18:
Spoke too soon, that fixed one error but there are more.  So much for trying to multitask to quickly fix this.  I'll look at this more tomorrow.

nvisible, 2016-04-18:
Thanks for the quick response! If there's any testing you need me to do I'll be your Huckleberry.

embedded-creations, 2016-04-19:
OK, one more fix committed, I can now compile and run sketches in Arduino 1.6.8 with Teensyduino 1.28.  I'll package this up into a release ASAP, hopefully today.

embedded-creations, 2016-04-19:
New release is done.  @nvisible please LMK if it's working for you when you have time

nvisible, 2016-04-21:
It compiles like a charm! Thanks for this!

closed 2016-04-21 by nvisible