pixelmatix/SmartMatrix #38: How to create additional indexed layers?

nvisible, 2016-04-30:
Is there a way to create additional indexed layers in a sketch? Or even multiple background layers? I'm building an EV dashboard and I'm using the colors of variables to also indicate their value. Thanks.

nvisible, 2016-04-30:
You know, I may have found my answer in the allocate section....

nvisible, 2016-04-30:
Sorry, been coding too early in the morning. I'm more hardware guy than software. For any other noobs like me that may be reading this in the future here's my train of thought:

> 
You know Arduino functions like Serial.begin()? 'Serial' is the name and 'begin' is the command. Well when I saw indexedLayer.drawChar I'm thinking indexedLayer is baked in. But it's not. You set the name of the layer you want by allocating it in the top section like in the FeatureDemo sketch.

closed 2016-04-30 by nvisible

embedded-creations, 2016-04-30:
The (lack of) documentation for SmartMatrix 3.0 could use some work.  Glad you got it figured out!