pixelmatix/SmartMatrix #42: DMAChannel.h missing?

bsenglert, 2017-04-19:
I looked for a DMAChannel library and could not find one. Does it come from outside of the SmartMatrix library folder?

In file included from C:\Users\Englert\Documents\Arduino\libraries\SmartMatrix3\src/SmartMatrix3.h:167:0,

                 from C:\Users\Englert\AppData\Local\Temp\arduino_modified_sketch_327038\SmartMatrix.ino:1:

C:\Users\Englert\Documents\Arduino\libraries\SmartMatrix3\src/SmartMatrix_Impl.h:26:24: fatal error: DMAChannel.h: No such file or directory

 #include "DMAChannel.h"

                        ^

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.
.

embedded-creations, 2017-04-19:
You're compiling for the Arduino Mega, but this library is for the Teensy 3.  DMAChannel.h is included by the Teensyduino add-on for Arduino, and only applies to the Teensy hardware.

closed 2017-04-19 by embedded-creations

bsenglert, 2017-04-19:
very good, I think I understand now. Thanks so much for the response.