pixelmatix/SmartMatrix #44: Led panel 192 x 64

pinballsp, 2017-06-07:
Hello:

It's possible config SmartMatrix to work with a Led panel 192 x 64 pixel ?,  I do this Led panel with 3 x 64*64 pixel led panels, 1/32 Scan, HUB75E.

Kind Regards

embedded-creations, 2017-06-07:
It might work with the Teensy 3.5 or 3.6, which has more CPU speed and RAM, but you'll probably have to drop down the color depth.  The Teensy 3.2 can handle about 64x64 max.  For a huge panel like that you might want to look at solutions for the Raspberry Pi: https://github.com/hzeller/rpi-rgb-led-matrix

pinballsp, 2017-06-07:
ok, thank you.
I have think export Smartmatrix code to Atmel SAM S70, ARM Cortex M7, 300Mhz. Both Kinetis and SAM S70 are ARM, so hope its not very hard translate source code.

Now I run Smartmatrix in a Kinetis MK66 at 180Mhz, same that install Teensy 3.6, I develop my own PCB boards.

embedded-creations, 2017-06-07:
The SmartMatrix Library uses DMA hardware that's very specific to the Kinetis MK66.  I investigating porting it to the processor in the Teensy LC, and the STM32 processor that's in the Particle Photon, and found it to be too hard to port.

I'm simplifying the DMA by adding some hardware that will be in the upcoming SmartMatrix Shield V4.  Keep an eye out for the hardware and updated library in the next month or so.  That version will be more portable, it only uses 2 DMA channels instead of 4.  It still won't be that easy to port.

pinballsp, 2017-06-07:
Ok, fine, thank you, I will check for new update.

embedded-creations, 2018-02-17:
Works with Teensy 3.6 and some modifications I'm about to commit:
http://community.pixelmatix.com/t/modify-code-for-64-64-pixel-hub75e-1-32-scan-led-panels/212/13?u=louis

closed 2018-02-17 by embedded-creations