pixelmatix/SmartMatrix #47: Try improved font code

embedded-creations, 2017-11-28:
From Álvaro Monares G.

> I had improve the fonts using fonts from any size (width and height), but there is at least a backward compatibility. For instance getBitmapFontRowAtXY I add the X value.
I have try this on a panel 64(width)x32(height) and works, now I don't have the panels so can't continue testing.

> It seem to work ok, I also change the location variable at getBitmapFontLocation on MatrixFont.cpp, it was static I remove the static keyword.

[SmartMatrix-src.zip](https://github.com/pixelmatix/SmartMatrix/files/1511795/SmartMatrix-src.zip)

amonares, 2017-11-29:
It supose that you can use in this way:
```c
backgroundLayer.setBitmapFont(font);
backgroundLayer.drawString(2, 1, {12, 242, 183}, "SmartMatrix");
```
Also you can use a function to get the size of the string in pixels and so locate at the center of the matrix
```c
backgroundLayer.drawString(kMatrixWidth / 2 - getBitmapFontStringWidth(charBuffer, &font) / 2, TOP_OFFSET, {0, 0, 0xff}, charBuffer);
```

Example font
```c
// Created from bdf2c Version 3, (c) 2009, 2010 by Lutz Sammer
//	License AGPLv3: GNU Affero General Public License version 3

#include "MatrixFontCommon.h"

	/// character bitmap for each encoding
static const unsigned char __font_bitmap__[] = {
//  32 $20 '0020'
//	width 4, bbx 0, bby 0, bbw 0, bbh 0
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
//  33 $21 '0021'
//	width 6, bbx 3, bby 0, bbw 1, bbh 13
	___X____,________,________,
	___X____,________,________,
	___X____,________,________,
	___X____,________,________,
	___X____,________,________,
	___X____,________,________,
	___X____,________,________,
	___X____,________,________,
	___X____,________,________,
	___X____,________,________,
	________,________,________,
	___X____,________,________,
	___X____,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
	________,________,________,
//  34 $22 '0022'
//	width 8, bbx 1, bby 8, bbw 5, bbh 5
	_XX__X__,________,________,
	_XX__X__,________,________,
	_XX__X__,________,________,
	_XX__X__,________,________,
	_XX__X__,________,________,
[... 4527 lines cut ...]
	242,
	243,
	244,
	245,
	246,
	247,
	248,
	249,
	250,
	251,
	252,
	253,
	254,
	255,
};

	/// bitmap font structure
const struct bitmap_font font = {
	.Width = 17, .Height = 20,
	.Chars = 191,
	.Widths = __font_widths__,
	.Index = __font_index__,
	.Bitmap = __font_bitmap__,
};
```

karakbot, 2019-05-09:
@amonares i tried this method, but its throwing me up with multiple errors. I am using 64x192 panels, first error while compiling is that "const uint8_t kPanelType = SMARTMATRIX_HUB75_64ROW_MOD32SCAN; " is not defined in this scope. Any thoughts ?

embedded-creations, 2020-11-23:
SmartMatrix Library 4.0 has reworked fonts using Adafruit_GFX, I'm closing this issue

closed 2020-11-23 by embedded-creations