pixelmatix/SmartMatrix #49: Add support for 64x64 panels with AB address lines

nbJosh, 2018-03-18:
there are 64x64 HUB75 panels that only have A & B address lines.   

The Raspberry pi LED driver project has added support for these panels,  I would like to drive the same panels with my Smart Matrix V4 board, 

see this commit as an example.

https://github.com/hzeller/rpi-rgb-led-matrix/commit/829d40e9ff00dd315b8315e5647088fbd4d74a7f

embedded-creations, 2018-03-18:
Interesting, I've never seen a 64x64/4 panel.  The only /4 panels I've only seen are 16x32/4 panels.  Do you have a link to where I could buy one?

Support for this is planned, see #34.  I have a project in mind that requires support for /4 so I'm hoping I get to this sooner than later.  It's not an easy addition though, as there's no support for panels that refresh multiple rows at once in the library.

nbJosh, 2018-03-19:
Here is the link on AliExpress 
http://s.aliexpress.com/n2UvUFfe

embedded-creations, 2018-03-22:
Thanks, I ordered a panel, and have other /4 panels here to test if I have time to work on this before the 64x64 panel arrives.

embedded-creations, 2018-04-09:
I ordered the panel you linked from Aliexpress, and instead of what was shown, received a /32 panel with ABCDE address lines.

I took another look at the Hzeller code and [this discussion](https://github.com/hzeller/rpi-rgb-led-matrix/issues/120) and realized that these panels aren't /4, they are /32 but use some method to shift in the address instead of directly driving.  This won't work with the SmartLED Shield hardware, as the address is meant to be only changed once per latch cycle.  These AB panels require a separate clock and data line to cycle 5 times per row to send the address, and the address lines used in the SmartLED Shield can only update once per row.

I unfortunately don't see a straightforward way to support these AB panels with the SmartLED Shield or the SmartMatrix Library.

closed 2018-04-09 by embedded-creations

marcmerlin, 2019-04-17:
PxMatrix may support those panels according to
https://github.com/2dom/PxMatrix/blob/master/PxMatrix.h#L261
and https://github.com/2dom/PxMatrix/blob/master/PxMatrix.h#L438
if anyone finds this bugs and still needs AB support (required by most P2 panels)

@embedded-creations  is the limitation you reported also valid on the ESP32 driver?

embedded-creations, 2019-04-29:
@marcmerlin 

> is the limitation you reported also valid on the ESP32 driver?

Now that I think about it, no it's not.  The ESP32 driver happens to use a transparent latch instead of a flip flop, so the AB signals can go through the latch, and the latch signal is controlled by GPIO not a timer, so it should be straightforward to toggle GPIOs to send the AB pattern needed.

I'm making support of AB panels a requirement for the APA102-HUB75 adapter project, so it's back on my radar, but I probably won't be working on it anytime soon.  If you want to work on it, I can help point you in the right direction on the forum.

marcmerlin, 2019-04-29:
Thanks for the details @embedded-creations . 
I luckily do not have any AB panels at this time, so I don't need this, but I pinged this bug while I was comparing libraries and their support. Given what you said, do you mind re-opening the bug then (it's marked as closed ATM).

For my own use, however, given the memory limitations I'm hitting past 96x64 on both teensy and ESP32, I'm currently looking into rPi solutions for bigger things, and they already support AB I believe.

marcmerlin, 2019-05-28:
For completeness sake, there are some yet new panels with ABC addressing for 64 lines (32 addresses, but really 8 and lines are shifted 4 times).
Details on this post https://community.pixelmatix.com/t/sigh-new-panels-abc-with-fm26126a/460/3
I don't expect support, but I'm listing this here along with the AB panels.

embedded-creations, 2019-10-30:
I'm reopening this issue as there's a possibility these panels can be supported with the ESP32 chipset.

I unfortunately still don't have any AB panels despite trying to purchase from sellers that shipped AB panels to other people.  If anyone has any leads on where to get one, or is willing to donate or sell me one (I live in the UK right now, but sometimes visit the US), please let me know.

marcmerlin, 2019-12-26:
As a quick update, ABC panels also seem to be AB like panels as explained in 
https://github.com/hzeller/rpi-rgb-led-matrix/issues/823#issuecomment-534003119
So, if you can't find AB panels, getting AC ones may be close enough

maikkel, 2021-02-28:
I bought this 128x128 panel on AliExpress (photos do not match):
https://www.aliexpress.com/item/1005001999259954.html?spm=a2g0s.9042311.0.0.27424c4dcPFl1x
Here my own photos from the panel i received:
https://imgur.com/a/CJO7y8w
If i understand right, this is an ABC-Panel, and it doesn't work. when plugging in while powered (i know, that's not a great idea), it shortly flickers half of the panel, but otherwise stays black, no matter which settings i choose (have a 64x64 panel running fine).
Would you be interested in this? I might be willing to sell it and send it over (I'm in Switzerland) since i have no use for it if it won't work.

marcmerlin, 2021-02-28:
Yes, you should try the rpi-rgb-panel driver on rPi, this is not supported by SmartMatrix.

marcmerlin, 2021-02-28:
To be honest, I don't understand the point of ABC panels, they save 2 IO pins, at the expense of speed.
Now, the other thing is that I've never seen 128x128 before. 
An ABCDE panel can only drive 128x64, so 128x128 may need some extra IO addressing over ABC (which then would really be ABCDEF, 6 address lines), or maybe it's configured as 256x64, hard to say.
Basically I'll correct what I said: I don't think this panel will be supposed by any open source driver out of the box, but maybe the rpi ABC driver can be easily changed to support 6 address lines instead of 5).

embedded-creations, 2022-11-28:
> Any chance that there will be support for teensy 4.x?

The wiki has details on support for these panels.  Short summary: it's not possible to support them with the Teensy, and it's possible to support them on the ESP32 though it's not something I plan to work on anytime soon.

https://github.com/pixelmatix/SmartMatrix/wiki/HUB75-Panels#abacabc-panels