pixelmatix/SmartMatrix #50: Cannot use ESP32 GPIO0 in I2S bus?

paulreimer, 2018-05-22:
I had already made a board, with my own I2S parallel HUB75 driver, and my board also works with ESP_Sprite's code/driver.

However, it seems I cannot use GPIO0 for any of the I2S signals (my board uses GPIO0<->OE) with SmartMatrix `teensylc` branch. All other signals seem fine on a logic analyzer (I tested re-assigning to another pin and it looked like the correct signal.) It just won't work on GPIO0 (outputs just a logic high the whole time)

embedded-creations, 2018-05-22:
It's working fine for me, sounds like a hardware issue.  (The SmartLED Shield circuit for ESP32 is using GPIO for BUFFER_B2.) GPIO0 is typically used for a button input on ESP32 dev boards, is the button circuit interfering?  Can you toggle GPIO0 with a normal sketch, e.g. the Arduino Blink sketch with the pin changed?

embedded-creations, 2018-05-22:
Can you post your modified `MatrixHardware_ESP32_V0.h` file?

paulreimer, 2018-05-22:
Sure, here it is: (I removed the alternate pinouts, and my board does not have a latch -- although, good idea, I will use one next time to free up some pins)

```
// LICENSE
// Note: only one MatrixHardware_*.h file should be included per project

#ifndef MATRIX_HARDWARE_H
#define MATRIX_HARDWARE_H
// formula used is 80000000L/(cfg->clkspeed_hz + 1), must result in >=2.  Acceptable values 26.67MHz, 20MHz, 16MHz, 13.34MHz...
#define ESP32_I2S_CLOCK_SPEED (20000000UL)

// ADDX is output directly using GPIO
#define CLKS_DURING_LATCH   0

//Upper half RGB
#define BIT_R1  (1<<0)
#define BIT_G1  (1<<1)
#define BIT_B1  (1<<2)
//Lower half RGB
#define BIT_R2  (1<<3)
#define BIT_G2  (1<<4)
#define BIT_B2  (1<<5)

// Control Signals
#define BIT_LAT (1<<6)
#define BIT_OE  (1<<7)

#define BIT_A   (1<<8)
#define BIT_B   (1<<9)
#define BIT_C   (1<<10)
#define BIT_D   (1<<11)
#define BIT_E   (1<<12)

#define R1_PIN  33
#define G1_PIN  25
#define B1_PIN  26
#define R2_PIN  27
#define G2_PIN  2
#define B2_PIN  5

#define A_PIN   16
#define B_PIN   17
#define C_PIN   18
#define D_PIN   19
#define E_PIN   12
#define LAT_PIN 21
#define OE_PIN  0

#define CLK_PIN 4

#define GPIO_PWM0A_OUT GPIO_NUM_12
#define GPIO_SYNC0_IN  GPIO_NUM_34

//#define DEBUG_PINS_ENABLED
#define DEBUG_1_GPIO    GPIO_NUM_13
//#define DEBUG_2_GPIO    GPIO_NUM_12

#endif
```

Also, I am using this in esp-idf / FreeRTOS (no Arduino) so I did have to make a few changes to pass compiler warnings (`extern "C"` some things, include some FreeRTOS headers) and defined `millis()` as:
```
unsigned long millis()
{
  return (esp_timer_get_time() / 1000);
}
```

paulreimer, 2018-05-22:
I also tried with a few combinations of OE_PIN and GPIO_PWM0A_OUT settings (i.e. both assigned to GPIO0, or just OE_PIN, or just GPIO_PWM0A_OUT), as well as commenting out the `setupTimer()` function.

embedded-creations, 2018-05-22:
Hmm, I'm stumped, my suggestion was going to be to comment out `setupTimer()` as that functionality isn't used yet, and probably won't work with your circuit anyway, but you already did that...

embedded-creations, 2018-05-22:
It could have to do with using esp-idf, though I don't know why exactly.  Can you try just toggling that pin with SmartMatrix Library running and see if that works?  Try with both `OE_PIN` set to 0 and set to something else.

embedded-creations, 2018-05-22:
Did you try running the ESP_Sprite's example code with my modifications?

paulreimer, 2018-05-22:
Got it working! My bad -- there was some code in one of my components (that was copied when I made this test app) using GPIO0 already. Completely my fault, removing that code makes it work just fine.

closed 2018-05-22 by paulreimer

embedded-creations, 2018-05-23:
Glad you got it working.

The extra bit from MCPWM OE control isn't working yet.  I did some tests to make sure I could get an extra bit (btw it could be more than one bit if the I2S clock is slowed down even further), and you can see that code here: https://github.com/pixelmatix/SmartMatrix/blob/teensylc/src/SmartMatrixMultiplexedCalcEsp32_Impl.h#L449

You can follow the circuit that's in the SmartLED Shield schematic to eventually get the extra bit.  That extra line separated by a jumper is not needed.  You'll need one extra line to output the OEPWM signal, and that can be moved around by the #define.  Change `GPIO_SYNC0_IN` to match your OE output, I'm guessing `GPIO_NUM_0`.  Change `GPIO_PWM0A_OUT` a free output capable pin.  With the MCPWM code enabled you should be able to see pulses on the `GPIO_PWM0A_OUT` that are synced with the I2S OE output, even though the SmartMatrix Library isn't taking advantage of them.  You'll need the latch and the MUX circuit on your board to switch between OE_DMA and OE_PWM.

Sure, the PR is welcome.  I'll just need to make sure it doesn't interfere with the ESP32 or Teensy Arduino functionality.

nbJosh, 2018-05-25:
I have been experimenting with several ESP32 examples driving my 64x64 rib panel.   I have a board put together that successfully drives my panel.  However when I setup the same pins for these I2S examp,es,  the result is very scrambled, with the fonts rendered backwards, and sliced up 
[image]

Any idea where to look?

Again,  this same hardware works fine with the non I2S RGB panel driver
Also,  it is a completely passive board

embedded-creations, 2018-05-25:
You should probably open up a new issue, either here or on community.pixelmatix.com

The SmartMatrix Library is running at 20MHz clock, which is probably too fast for a circuit without buffers.  Look in `MatrixHardware_ESP32_V0.h` for where to change it

Maybe there's a wiring issue explaining the backwards text.  I don't understand why it would be backwards, but maybe it's actually inverted top to bottom and your panel is rotated

masrodjie, 2019-03-17:
I have same problem with @nbJosh and trying to lowering clock to 1MHz didn't like @embedded-creations suggest didn't solve the probem.

embedded-creations, 2019-03-19:
@masrodjie You should probably open up a new issue, either here or on community.pixelmatix.com, this issue is for something unrelated and is now closed