pixelmatix/SmartMatrix #51: ESP32: Refresh fails at higher display sizes: out of memory

cmair, 2018-05-30:
Hi,

I found that the library fails at higher panel sizes because
`matrixUpdateFrames = (frameStruct *)heap_caps_malloc(sizeof(frameStruct) * ESP32_NUM_FRAME_BUFFERS, MALLOC_CAP_DMA);`
returns 0 and therefore the DMA engine does not start sending data.

There are a few other places where assert() is used after malloc without checking the return value.

Optimizing some other tasks (memory-wise) allows me to drive a 64x96 matrix with RGB24. 64x128 still does not work :(

embedded-creations, 2018-06-04:
Thanks for reporting the issue, I'll get that fixed.

I've been trying to support larger panels as well, and I think the next step is for me to break up the single frame buffers malloc into two separate malloc calls which should allow for larger matrix sizes where there's enough memory, but not enough all in a single region:

https://github.com/pixelmatix/SmartMatrix/blob/teensylc/src/SmartMatrixMultiplexedRefreshEsp32_Impl.h#L190

embedded-creations, 2018-06-04:
I did try 64x128 myself, and saw it working with the sketch I was using at the time (I don't recall which one, maybe Bitmaps).  128x128 didn't work.  Try 64x128 with 24-bit color instead of 36-bit color

embedded-creations, 2018-06-21:
I just pushed some improvements that split that large malloc in two, and fail on malloc errors more gracefully.  I tried 96x128/24-bit with the Bitmaps example and it worked.  Hopefully your sketch works now, please let me know.

closed 2018-06-21 by embedded-creations