pixelmatix/SmartMatrix #60: Smartmatrix shield v4 and Teensey 3.6 not working on 64x32 display

dhpete, 2018-12-19:
Hey, guys. I bought some 64x32 panels from amazon, got a teensey 3.6 and smartshield v4, hooked everything up( I think, the correct way) and nothing. Plenty of power, code compiles( using the Feature Demo sketch) Loader seems to program, but display does not come on. Without the Teensey/Shield installed, you can touch the pins on the matrix and get a reaction, so I dont think its DOA. Even tried 2 different panels with the same results. Is there something I am missing? Sorry if its a dumb question, but Im a total NOOB to this matrix thing. Any help would be great.

embedded-creations, 2018-12-19:
Did you remove the comment at the top of the sketch to tell the library you're using SmartLED Shield V4?  (This is done by default now but I don't know what version you're using)

It should look like this at the top:
```
#include <SmartLEDShieldV4.h>  // comment out this line for if you're not using SmartLED Shield V4 hardware (this line needs to be before #include <SmartMatrix3.h>)
#include <SmartMatrix3.h>
```

dhpete, 2018-12-19:
Thanks, Louis. Mine was uncommented. After pulling the shield (while it was on) I now get a 16x16 green section in the lower right corner. Well, its a start :). If I upload the FastLED functions sketch, the display goes off, reload the other sketch and back to the green spot. Sorry for bothering you guys...this might take me a while.

embedded-creations, 2018-12-19:
Can you upload a picture of your your shield, wiring, power?

Can you take a picture of the back of one of the panels showing any pinout diagrams that are on the board?

dhpete, 2018-12-19:
[image]

[image]

[image]

[image]

[image]

Sorry for the crappy pics. Power is from a 5v 60a power supply. 2 64x32 panels but only one connected. Thought I would get one working, then go through the process for 64x64. No labels on the pin outs but I went by this https://cdn-learn.adafruit.com/downloads/pdf/32x16-32x32-rgb-led-matrix.pdf

embedded-creations, 2018-12-19:
Try running this precompiled sketch.  Unzip, drag the .hex into the TeensyLoader window, and make sure the name in the lower section of TeensyLoader updates, then press the button on the Teensy 3.6 and make sure it loads.

[image]

[FeatureDemo-Teensy3.6-32x32-V4.ino.hex.zip](https://github.com/pixelmatix/SmartMatrix/files/2695649/FeatureDemo-Teensy3.6-32x32-V4.ino.hex.zip)

dhpete, 2018-12-19:
This is what I got
[image]
The diagonal lines are from my camera...not visible to the eye

marcmerlin, 2018-12-19:
@dhpete  just to confirm, I bought a 64x32 panel from amazon like yours, plugged in a teensy 3.6 into the same shield you bought, and it just worked. Louis is already helping you, but just confirming that this exact setup should work fine.
As a slight note, usually it's better to ask for help on a forum ( https://community.pixelmatix.com ) than open an issue unless you're pretty sure that you found an issue with the library.

dhpete, 2018-12-21:
About to just call this project a bust. Searched the forums and tried a few things, but still nothing. At least the clock module and remote will be here today (sarcasm). Seems the chances of getting 2 bad panels is low( although possible) Teensey seems to work for other applications and mounted to the shield, all GPIO seem to function. Im at a loss. Was looking forward to running Aurora on this. Thanks for all the help, just not sure where to go from here.

dhpete, 2018-12-21:
> I just read again that you tested the GPIO, sounds like it’s a problem
> with the shield. Sorry about that. Please post back after you get the
> replacement and let us know if it works

Thanks. Tested GPIO on Teensey, and GPIO on the shield with Teensey on
board. Not sure how to rest the ribbon/matrix connector output, but seems
like all pins on the board work. Tried one panel with a Arduino Mega using
the RGB matrix panel sketch and couldnt get that to work either. Could be
my code there, or I have 2 bad panels. Im not too great at this
troubleshooting. So far my Arduino/Pi experience has been you plug it in,
and it works. Thanks for all your help, just wish I could get this up and
running as I really wanted to use Aurora.

embedded-creations, 2018-12-22:
> So far my Arduino/Pi experience has been you plug it in, and it works

Yes, that's the goal of the SmartLED Shield.

Have you tried connecting both with directly plugging in the SmartLED Shield to the panel, and with using the pins and ribbon cable to connect between shield and panel?

How did you test the GPIO signals going through the SmartLED Shield?  There are level shifting buffers on the shield, and if one of those had a loose connection, the signals may not be working properly on the HUB75 connector side.  A good test you could do with just a multimeter is to have the Teensy toggle the pins between 0/1 at a slow rate, so you can easily see them change with the multimeter.  Plug the male-male pins into the HUB75 connector on the shield so you can probe them.  Toggle pins and look for the change with the multimeter.  Especially look for Teensy pin 3 to toggle the "LT" signal on the HUB75 connector, and Teensy pin 14 to toggle the "CK" signal as that's where I think the issues could be based on your troubleshooting so far.

Sorry for the continued hassle with this.

embedded-creations, 2018-12-22:
And if you'd rather just get a replacement board instead of continuing to troubleshoot, as I said before I encourage you to request a replacement Shield from where you bought your SmartLED Shield.

closed 2020-11-23 by embedded-creations