pixelmatix/SmartMatrix #65: SmartMatrix + FastLED (Neopixels) ESP32 crash

marcmerlin, 2018-12-30:
When using NeoPixels on ESP32 with  https://github.com/samguyer/FastLED
everything works fine.
When using https://github.com/FastLED/FastLED instead, SmartMatrix crashes.

Normally I'd blame FastLED but the crash is in SmartMatrix. I'm not very familiar with ESP32 crashes/panics. Are they usually correct about which library is at fault when both cores are used, or could the problem very well be a bug in FastLED master that is fixed in Sam Guyer's branch?

This is not a must fix for me, I'm fine using Sam's ESP32 branch which probably is better/more up to date, but I'm filing this bug in case it points to a problem in SmartMatrix.

Guru Meditation Error: Core 1 panic’ed (Interrupt wdt timeout on CPU1)
PC: 0x400d0f84: SMLayerBackground ::swapBuffers(bool) at /home/merlin/Arduino/libraries/SmartMatrix_ESP32/src/Layer_Background_Impl.h line 921
EXCVADDR: 0x00000000

Decoding stack results
0x400d0f84: SMLayerBackground ::swapBuffers(bool) at /home/merlin/Arduino/libraries/SmartMatrix_ESP32/src/Layer_Background_Impl.h line 921
0x400d1c3c: loop() at /home/merlin/arduino/libraries/SmartMatrix_ESP32/examples/DemoFastLED/DemoFastLED.ino line 163
0x400e7806: loopTask(void*) at /home/merlin/Arduino/hardware/espressif/esp32/cores/esp32/main.cpp line 17

Code:
https://github.com/marcmerlin/SmartMatrix/blob/teensylc/examples/DemoFastLED/DemoFastLED.ino

AndreasMainz, 2020-08-29:
You got a watchdog error. Just try to fix it with delay(10) in the Loop function.
It helped in my case..

embedded-creations, 2020-11-23:
the calcTask on ESP32 now yields to reset the watchdog (in the upcoming SmartLED Shield 4.0 release), I hope this is resolved:

```
            if(currentMillis - lastMillis >= 4500){
                // sleep a bit to reset the watchdog (default is 5000ms between resets)
                vTaskDelay(1);
                lastMillis = currentMillis;
            }
```

closed 2020-11-23 by embedded-creations