pixelmatix/SmartMatrix #69: Update AnimatedGIFs sketch for ESP32 to use malloc and avoid compile/startup issues with large buffers

embedded-creations, 2019-03-10:
https://github.com/pixelmatix/AnimatedGIFs/issues/26

marcmerlin, 2020-05-31:
if anyone needs this it's in my fork: https://github.com/marcmerlin/AnimatedGIFs

marcmerlin, 2020-11-27:
if it still uses arrays instead of mallocs, the problem isn't fixed.
arrays get alloced early in a smaller pool in ESP32, and just don't fit on a decent size display, but they fit if you use malloc because that goes into another pool (yeah, ESP32 sucks in that way)

embedded-creations, 2020-11-27:
OK, thanks for the clarification, I'll reopen it.

embedded-creations, 2020-11-27:
I think this shows how to allocate the block needed for AnimatedGIF with malloc (after allocating mymemorypool with malloc):

https://github.com/bitbank2/AnimatedGIF/issues/8#issuecomment-676480432

marcmerlin, 2020-11-27:
I'm only seeing the use of a single static array. If so, that is not as good as malloc as ESP32 has memory available at runtime with malloc but not available in pre-allocated buffers (global arrays).
Also, to use psram if you choose to, you have to use malloc.

This is what I did to my old fork of the library, I switched everything to malloc.
Sorry, this is not a single CL, I wrote it over time
- https://github.com/marcmerlin/AnimatedGIFs/commit/0c6191181ec3a6b7061afcb9acf1512a7a19a358#diff-1c6b5b8854f2713658794a513e0149a7861b1aae2755e52dbb1c78a1900b2f2b
- https://github.com/marcmerlin/AnimatedGIFs/commit/2715fe2fccc666d667ad5932c75414d0cbe875e6#diff-1c6b5b8854f2713658794a513e0149a7861b1aae2755e52dbb1c78a1900b2f2b
- https://github.com/marcmerlin/AnimatedGIFs/commit/c2dc4c78d3fe8d0cc45d8aa621457466a48d4713#diff-cca9dc0aaf37dcf8e83c111f403400c9f29a228fbfad3c5f62eed507f8608664
- https://github.com/marcmerlin/AnimatedGIFs/commit/1750be0fb5062c8f29f4535e7e08e60036746032#diff-cca9dc0aaf37dcf8e83c111f403400c9f29a228fbfad3c5f62eed507f8608664
- https://github.com/marcmerlin/AnimatedGIFs/commit/ad21ba5522d75cfd02fe517a081fbb15eb4de1a9#diff-cca9dc0aaf37dcf8e83c111f403400c9f29a228fbfad3c5f62eed507f8608664

I also later updated mallocordie to allow allocating in PSRAM to further save main memory

marcmerlin, 2020-11-27:
mallocordie might be helpful to you: https://github.com/marcmerlin/FastLED_NeoMatrix_SmartMatrix_LEDMatrix_GFX_Demos/blob/38a137e3ac54d4f0b8d7a11ca99d2061cbbf86b1/neomatrix_config.h#L885

embedded-creations, 2020-11-27:
Thanks for the links, I'm not working on this now, but they look useful for later