pixelmatix/SmartMatrix #70: consider renaming teensylc branch?

marcmerlin, 2019-03-28:
Given what the branch is today, maybe rename it smartmatrixng ?
It's not very obvious to use a teensylc branch for esp32, especially if the branch doesn't work on teensylc (not sure about that part)

embedded-creations, 2019-03-31:
Good idea, but what about all the links that are already pointing at the GitHub /teensylc branch?

Thoughts:

- branch (not rename branch) from `teensylc` and name it `esp32` or whatever, make final commit to delete the contents of the `teensylc` branch except for a simple README.md saying "use the new `esp32` branch"

marcmerlin, 2019-03-31:
Your suggestion is much better

rorosaurus, 2019-10-09:
That makes sense to me.
Piggybacking off this, is there a reason why we want to keep ESP32 compatibility out of master branch?  Can we enumerate the work that is needed?

embedded-creations, 2019-10-11:
> is there a reason why we want to keep ESP32 compatibility out of master branch?

That branch is a huge refactoring of the code, not just ESP32 but also Teensy.  It's only alpha quality, and some things are known to be broken, other issues haven't been exposed yet because of not enough testing.

> Can we enumerate the work that is needed?

There's quite a bit of work listed in the README

closed 2020-11-23 by embedded-creations