pixelmatix/SmartMatrix #77: Will this library work with ESPAsyncWebServer library ?

GeorgeFlorian, 2019-04-12:
Hello !

OS: Linux Mint 17.1
Board: ESP32 Wrover-B DevKitV4
Display: [2727 SMD P5 64x32](https://www.aliexpress.com/item/SMD-2727-P5-outdoor-Video-wall-led-module-high-brightness-5500-nits-1-8-scan-64/32788680216.html)

I am having trouble finding a LED library that will work with [me-no-dev's ESPAsyncWebServer library](https://github.com/me-no-dev/ESPAsyncWebServer).
I believe that library also accesses RAM. And this library also accesses RAM.

I've tried [PxMatrix](https://github.com/2dom/PxMatrix) and I wasn't able to make it work with ESPAsyncWebServer.

Do you have any idea if this will work ?

Thank you !

GeorgeFlorian, 2019-04-16:
My project project uses SPIFFS + ESPAsyncWebServer to serve some web-pages while in AP Mode and some other web-pages while in STA Mode.

The problem is that the [PxMatrix](https://github.com/2dom/PxMatrix) is not compatible with the SPIFFS file system.

The LED Library uses the following methods to work:

```
void IRAM_ATTR display_updater(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  display.display(display_draw_time);
  portEXIT_CRITICAL_ISR(&timerMux);
}

void display_update_enable(bool is_enable){
  if (is_enable) {
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &display_updater, true);
    timerAlarmWrite(timer, 2000, true);
    timerAlarmEnable(timer);
  }
  else  {
    timerDetachInterrupt(timer);
    timerAlarmDisable(timer);
  }
}
```

I need to close the display when manipulating a SPIFFS file and then open it. So let's say: 
```
  server.on("/url", HTTP_GET, [](AsyncWebServerRequest *request){
    display_update_enable(false);
    request->send(SPIFFS, "/url_page.html", "text/html");
  });
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    display_update_enable(false);
    request->redirect("/url");
  });
```

That should work.

But I have a `HTTP Client` inside  `void loop()` that constantly GETs the body of a URL. The URL is stored in a txt file inside the ESP and I need SPIFFS to access it. So, again, I need to close the display before `SPIFFS.open()` and open it after I finish with the SPIFFS.

```
void loop() {
  display_update_enable(false);
  if (WiFi.status() == WL_CONNECTED) {  //Check the current connection status 
    File urlRead = SPIFFS.open("/urlFile.txt");
    if(!urlRead) logOutput((String)"ERROR_INSIDE_loop ! Couldn't open file to read !");

    if(urlRead.size() > 10) {
      HTTPClient http;
      String URL;
      logOutput((String)"URL: ");
      fileReadLines(urlRead, &URL);
      http.begin(URL); //Specify the URL
      urlRead.close();
      display_update_enable(true);
      int httpCode = http.GET(); //Make the request
  
      if (httpCode > 0) { //Check for the returning code
        logOutput((String)"[HTTP] GET... code: " + httpCode);
        if(httpCode == HTTP_CODE_OK) {
            String payload = http.getString();
            logOutput(payload);
            getPlaces(payload);
          }
      } else {
          logOutput((String)"[HTTP] GET... failed, error: " + http.errorToString(httpCode).c_str());
        }  
      http.end(); //Free the resources
      } else {
        logOutput("URL Link is invalid ! Please enter another URL");
      }
  }
  display.clearDisplay();
  display.print(payload);
  delay(3000);
}
```

Again, this is good and all, but the problem is that the `ESPAsyncWebServer` is always on.
I can access a web-page while `void loop` is going like crazy and I have no means to know when I access a web-page.

This means that the display can be on while I try to access a web-page that's stored in SPIFFS.
Trying to access `http://*myIP*` or `http://*myIP*/url` sometimes work, sometimes it returns A LOT of Guru Meditation Errors.
If I access one of the respective web-page it's like walking on thin ice, because if I refresh it, it can always break.

PxMatrix relies heavily on `IRAM_ATTR` functions and interrupts.

Will this library be able to function together with SPIFFS, ESPAsyncWebServer and HTTPClient ?

Thank you and sorry for the long post !

embedded-creations, 2019-04-16:
I don't know about PxMatrix or ESPAsyncWebServer, sorry.

Marc Merlin runs his AnimatedGIFs sketch with SPIFFS, and I think found some issues with it, but worked around them.  Take a look at this: https://community.pixelmatix.com/t/new-animatedgifs-with-ffat-support-on-esp32/409

GeorgeFlorian, 2019-04-17:
> I don't know about PxMatrix or ESPAsyncWebServer, sorry.
> 
> Marc Merlin runs his AnimatedGIFs sketch with SPIFFS, and I think found some issues with it, but worked around them. Take a look at this: https://community.pixelmatix.com/t/new-animatedgifs-with-ffat-support-on-esp32/409

I would still love to give it a shot, since it looks like the most supported and professional library out of the ones I've already tried.
The problem is that I couldn't find any instructions on how to do the wiring.
Is there any: "**This is how you do it, step-by-step guide**" ?

GeorgeFlorian, 2019-05-03:
I've managed to do the wiring and to make my own mapping for a Display.

closed 2019-05-03 by GeorgeFlorian

embedded-creations, 2019-05-09:
Sweet, did you have to do anything special to get ESPAsyncWebServer to be compatible with SmartMatrix Library?

GeorgeFlorian, 2019-05-09:
> Sweet, did you have to do anything special to get ESPAsyncWebServer to be compatible with SmartMatrix Library?

Nope. It works perfectly fine with ESPAsyncWebServer, HTTP Client, SPIFFS and WiFi.