pixelmatix/SmartMatrix #81: FastLED_Functions example has seam in noise pattern on 32x64 panel

embedded-creations, 2019-05-10:
Should FastLED_Functions example calculate noise based on height/width in a different way?

There seems to be a seam at 32 pixels wide when driving a 32x64 panel

sutaburosu, 2021-01-19:
Perhaps it would be useful to describe the issue in hand.  The crux of the problem is [this line](https://github.com/pixelmatix/SmartMatrix/blob/master/examples/FastLed_Functions/FastLed_Functions.ino#L118). `char noise[][]` has the same dimensions as the matrix layout.  That line indexes into noise[][] with both [x][y] and [y][x].  Clearly, this will never work for non-square layouts.

I can't fathom whether or not FastLED's noise function is periodic.  I was hoping some modulus operation might be an efficient way to deal with rectangular layouts.  Instead, I chose to simply grow noise[][] to be large enough for either axis.

Things I've learned since opening this PR:
 * there are three sketches in this repo using a similar pattern in `fillnoise8()`
 * one of them, FastLed+APA102, [was already fixed on initial commit](https://github.com/pixelmatix/SmartMatrix/blob/master/examples/FastLed_Panel_Plus_Apa102/FastLed_Panel_Plus_Apa102.ino#L87-L102), but in a different style than my proposed fix.

I see at least 3 possible futures.  I'm not experienced enough in C/C++/SmartMatrix to have a worthwhile opinion on style, so it falls to you to choose between them, or something else entirely.  Personally, I spray `#define`s everywhere.  I'm aware that this is anathema to others.  Regardless of your preference,  I'm happy to press the appropriate buttons to get this PR into a mergeable state.
 1. fix the two problematic sketches with `const uint`, in the style of the current PR.
 2. change all three affected sketches to use `const uint`.
 3. fix the two problematic sketches with a `#define MAX_DIMENSION_OVERALL`, to match the style of FastLED+APA102.  Initially, I had a similar name, but then I remembered `INT_MAX` et al, and decided that MAX shouldn't be in the name, hence `kMatrixLongestSide`.

embedded-creations, 2021-01-19:
Thanks for your PR and nice comments about the project!

My preference is 2, I'd be happy with 1 if you don't want to go through the extra effot.