pixelmatix/SmartMatrix #84: Problems driving a big panel (would like 128x128) with ESP32

GilDev, 2019-05-29:
Hello,

First of all thank you for your work on this, I hope development will continue especially for the ESP32 platform.
I’m currently trying to drive a 2x4 grid of 64x32 panels with an ESP32 which adds up to a resolution of 128x128.
I’m currently tinkering with the FeatureDemo example and it works up to 3 panels. Adding one more makes the ESP32 crash:

```
Rebooting...
ets Jun  8 2016 00:22:57
 
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5860
entry 0x40080698
 
Starting SmartMatrix Mallocs
Heap/32-bit Memory Available: 377416 bytes total, 126216 bytes largest free block
8-bit/DMA Memory Available  : 289856 bytes total, 126216 bytes largest free block
SmartMatrix Layers Allocated from Heap:
Heap/32-bit Memory Available: 375912 bytes total, 126216 bytes largest free block
Starting SmartMatrix DMA Mallocs
sizeof framestruct: 00000000
DMA Memory Available before ptr1 alloc: 288288 bytes total, 126216 bytes largest free block
matrixUpdateFrames[0] pointer: 3FFB8B88
DMA Memory Available before ptr2 alloc: 288288 bytes total, 126216 bytes largest free block
matrixUpdateFrames[1] pointer: 3FFB8B98
Frame Structs Allocated from Heap:
Heap/32-bit Memory Available: 375848 bytes total, 126216 bytes largest free block
8-bit/DMA Memory Available  : 288288 bytes total, 126216 bytes largest free block
Allocating refresh buffer:
lsbMsbTransitionBit of 0 requires 49152 RAM, 126216 available, leaving 77064 free:
Raised lsbMsbTransitionBit to 0/7 to fit in RAM
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d1a17  PS      : 0x00060330  A0      : 0x800d1cf0  A1      : 0x3ffb1e80
A2      : 0x3b9aca00  A3      : 0x3ffbfd0c  A4      : 0x00000000  A5      : 0x0001ed08
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x800d1a11  A9      : 0x3ffb1e30
A10     : 0x00000031  A11     : 0x3ffb80c0  A12     : 0x00000007  A13     : 0x0001ed08
A14     : 0x00012d08  A15     : 0x3ffbfcd4  SAR     : 0x00000004  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff
 
Backtrace: 0x400d1a17:0x3ffb1e80 0x400d1ced:0x3ffb1f50 0x400d1d45:0x3ffb1f80 0x400d70c3:0x3ffb1fb0 0x400895a5:0x3ffb1fd0
 
Rebooting...
```

Here’s how I’ve setup the sketch (tried some C-shape stacking but saw it’s sadly currently broken):

```
#define COLOR_DEPTH 24                  // known working: 24, 48 - If the sketch uses type `rgb24` directly, COLOR_DEPTH must be 24
const uint8_t kMatrixWidth = 256;        // known working: 32, 64, 96, 128
const uint8_t kMatrixHeight = 32;       // known working: 16, 32, 48, 64
const uint8_t kRefreshDepth = 24;       // known working: 24, 36, 48
const uint8_t kDmaBufferRows = 2;       // known working: 2-4, use 2 to save memory, more to keep from dropping frames and automatically lowering refresh rate
const uint8_t kPanelType = SMARTMATRIX_HUB75_32ROW_MOD16SCAN;   // use SMARTMATRIX_HUB75_16ROW_MOD8SCAN for common 16x32 panels
```

If I set the matrix to 64x64 instead of 256x32, it works but sometimes crashes in the middle of drawing things and it just stays like that. Even with 3 panels it sometimes crashes when drawing things.

If I set the matrix to 128x128, of course it crashes, even with kDmaBufferRows set to 1:

```
Rebooting...
ets Jun  8 2016 00:22:57
 

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:9280
load:0x40080400,len:5860
entry 0x40080698
 
Starting SmartMatrix Mallocs
Heap/32-bit Memory Available: 371232 bytes total, 120340 bytes largest free block
8-bit/DMA Memory Available  : 283672 bytes total, 120340 bytes largest free block
SmartMatrix Layers Allocated from Heap:
Heap/32-bit Memory Available: 271412 bytes total, 113792 bytes largest free block
Starting SmartMatrix DMA Mallocs
assertion "matrixUpdateFrames[0] != NULL" failed: file "lib\SmartMatrix-teensylc\src/SmartMatrixMultiplexedRefreshEsp32_Impl.h", line 193, function: static void S
martMatrix3RefreshMultiplexed<refreshDepth, matrixWidth, matrixHeight, panelType, optionFlags>::begin(uint32_t) [with int refreshDepth = 24; int matrixWidth = 128
; int matrixHeight = 128; unsigned char panelType = 0u; unsigned char optionFlags = 0u; uint32_t = unsigned int]
abort() was called at PC 0x400da8eb on core 1
 
Backtrace: 0x40085484:0x3ffb1e10 0x400856b1:0x3ffb1e30 0x400da8eb:0x3ffb1e50 0x400d1812:0x3ffb1e80 0x400d1d31:0x3ffb1f50 0x400d1d89:0x3ffb1f80 0x400d70ff:0x3ffb1f
b0 0x400895a5:0x3ffb1fd0
 
Rebooting...
```

Do you please have any idea of how I can drive this big panel reliably? I don’t need huge refresh rate or FPS (brightness is more important). Maybe I’ve gotten to the limits of this library and need to check something else (maybe [with a Raspberry Pi](https://github.com/hzeller/rpi-rgb-led-matrix))? Thanks!

GeorgeFlorian, 2019-05-29:
Hello !
@marcmerlin pushed this library to 96x128. This means 12288 leds. More than that and it crashes. Also, the bigger the scan rate, the more memory you will need.
For more information please search the forum: https://community.pixelmatix.com/

You also need to consider that you will surely have other things that eat up RAM in your code: like Wi-Fi for example.

I personally used this library to drive one 64x32 panel and then two 64x32 panels in a 128x32 arrangement.
Then I've moved to Raspberry Pi and to the library you've already referenced: https://github.com/hzeller/rpi-rgb-led-matrix

GilDev, 2019-05-29:
Thanks for your answer and the confirmation about the informations I got.

I’ve seen indeed @marcmerlin’s discussion with Louis in [this topic](https://community.pixelmatix.com/t/cant-get-enough-dma-memory-on-esp32-assertion-matrixupdateframes-1-null/406), where it seems memory is quite a big problem. They talk about an external latch (also in the README) in order to cut in half memory usage but I haven’t fully understand that yet, need to make some researches.

One temporary solution could be to split my 128x128 panel in half and use two ESP32 with a serial connection to synchronize themselves.

Anyway, I’ll look into using a Raspberry Pi instead. @GeorgeFlorian Could I please know what’s the maximum resolution you were able to drive using the Raspberry Pi’s library?

GeorgeFlorian, 2019-05-29:
> Thanks for your answer and the confirmation about the informations I got.
>
> I’ve seen indeed @marcmerlin’s discussion with Louis in [this [...]

I've started using [the library](https://github.com/hzeller/rpi-rgb-led-matrix) with a Raspberry PI Zero, which is single-core and pretty slow and I've driven 4* 64x32 1/8 panels in a 128x64 arrangament.

You should be able to drive at least 36 panels in any arrangement.

I've ordered more panels and I am waiting for them ! I've also switched to a [Banana Pi 3](http://www.banana-pi.org/m3.html) which is way more powerful than the Pi Zero.

Here is what I am trying to achieve: https://github.com/hzeller/rpi-rgb-led-matrix/issues/819

TL; DR - I want to drive around 60 panels (6 wide, 10 height, so a 384x320 arrangement) using just the one Banana Pi 3. If I'm not able to use just one, that I most certainly will be able to achieve my goal using a second Banana Pi 3.

GilDev, 2019-05-29:
Great, thanks! I’ll get my Raspberry Pi 2 out of my drawer and experience with that. I will maybe make a boards with a [RPi CM3+](https://www.raspberrypi.org/products/compute-module-3-plus) in the future so it should be plenty enough for my needs.

GeorgeFlorian, 2019-05-29:
> Great, thanks! I’ll get my Raspberry Pi 2 out of my drawer and experience with that. I will maybe make a boards with a [RPi CM3+](https://www.raspberrypi.org/products/compute-module-3-plus) in the future so it should be plenty enough for my needs.

Great ! Good luck then !

closed 2020-11-23 by embedded-creations