pixelmatix/SmartMatrix #85: Color Correction appears to distort rainbow gradient

rorosaurus, 2019-06-30:
Hi all, thanks for this amazing library and the work to support ESP32! :)  
  
I noticed that a rainbow gradient I generate in FastLED is skewed, missing lots of yellow and orange.  I saw this mentioned as a risk in FastLED documentation, but I used the correct function which should account for that.  Eventually I realized I no longer see the issue if I disable color correction in Smartmatrix.  
  
Of course this distorts the other things I'm drawing in that layer, specifically the animated gif I'm rendering now appears very pale.  
  
Code: https://github.com/rorosaurus/FurretTotem/blob/851b317b4f9ea15884050d348c444b39de46d9be/src/FurretTotem/GifDecoder_Impl.h#L836  
Not-very-rainbowey-rainbow demo: https://gfycat.com/annualcookedheterodontosaurus

Perhaps one solution is to draw the GIF in another layer, which I will attempt tomorrow, but I anticipate running out of memory.  (128x64 1/32 panel)  My current hacky solution involves just drawing the rainbow when I detect a transparent pixel, to collapse everything into one layer.

However, in general I wouldn't expect the color correction to do this to the rainbow.  Am I missing something obvious?  Maybe I just don't understand color correction.  I notice if I disable color correction, then increase the gamma to compensate for the pale GIF, then I arrive at the same rainbow as the beginning: missing yellow and orange.

embedded-creations, 2019-07-03:
I've never seen any issues where color correction in SmartMatrix Library causes missing colors.

> saw this mentioned as a risk in FastLED documentation

Can you point me to where this is mentioned?

The image in this project was from a bitmap, but displayed with color correction on:

https://www.hackster.io/pixelmatix/smartmatrix-shield-for-teensy-3-1-8787f8

I looked at your code and don't see anything that jumps out as a problem, sorry.  Maybe it's just that the rainbow colors generated by FastLED only look rainbowey without color correction applied

It's not what you're asking, but I added an issue to improve handling transparent GIFs: https://github.com/pixelmatix/AnimatedGIFs/issues/39

rorosaurus, 2019-07-03:
It's warned in FastLED in the HSV to RGB conversion functions here: http://fastled.io/docs/3.1/hsv2rgb_8h_source.html
But I appear to be using that correctly.  FastLED itself also has color correction, and I can confirm the rainbow appears full there.

Right now I'm hoping to mitigate this by adding another layer for the background properly (hopefully I have enough RAM for that...) and disabling color correction for that layer.  I'll comment back here if I find anything else helpful towards this issue. :)

embedded-creations, 2019-07-04:
There's currently no support for a rgb24 layer (e.g. backgroundLayer) with transparency, so adding another layer for the background is going to be a big task.

embedded-creations, 2020-11-23:
I'm going to close this issue, but if there's still something you want me to look at with color correction, can you reproduce it in a simple sketch?

I think the issue is that you're using refresh depth of 24, and applying SmartMatrix's color (aka gamma) correction, and that's going to be lossy.  You don't see that with FastLED's color correction, because it's not doing the same thing.

That said, there still might be a bug to fix, but I haven't reproduced it here so I'm not working on it.

closed 2020-11-23 by embedded-creations

embedded-creations, 2020-11-24:
It's hard to tell as the codebase has changed a lot in the last year, but I applied a fix for a color correction bug that happens when you're using 24-bit color, so you could have been seeing that bug with your sketch

commit eb9eb35d (2020-11-24): Fix colorCorrection bug, thanks @ecurtz, fixes #31 and likely fixes #85