pixelmatix/SmartMatrix #87: USE SmartMatrix with P10 module

MohamadAlsadi, 2019-07-20:
hello my friend 

I am working on drive P10 module display ( 1 color)  via ESP32

I need library for drive 2 pcs from P10 module (64x16 display every one 32x16) 

please help me

MohamadAlsadi, 2019-07-20:
the link for ESP is 
https://www.shenzhen2u.com/NodeMCU-32S

I need show and scroll text 
please help me about ESP pins that must used and code for show and scroll text

embedded-creations, 2019-07-22:
The ESP32 port isn't finished and isn't well supported.  You can find some information on our forum here, and you can post questions there too if you don't see what you're looking for:

https://community.pixelmatix.com/t/smartmatrix-library-esp32-port/272/

closed 2019-07-22 by embedded-creations