pixelmatix/SmartMatrix #89: Any known issues with Teensy 4.0?

opentable-echamberlain, 2019-08-13:
Anyone try the library with the Teensy 4.0?

I’m hoping a faster board would have less flicker issues.

embedded-creations, 2019-08-14:
SmartMatrix Library doesn't support Teensy 4.0. From what I understand so far, this isn’t going to be an easy port, and Teensy 4.0 is very unlikely to be compatible with the SmartLED shield. I’m working on something that will allow driving HUB75 panels from any micro with an SPI port, so indirectly I’m working on Teensy 4.0 support, and that’s my priority for now.

I'd like to keep this discussion in one place and this thread in the SmartMatrix Community started first, but if you don't have an account feel free to post on GitHub instead:

https://community.pixelmatix.com/t/teensy-4-0-released/498

bleckers, 2020-03-05:
I know this isn't really a solution to your problem, but I've whipped together a rough Teensy 4.0/arbitrary pin driver library using the Adafruit libraries - https://github.com/bleckers/RGB-Matrix-Panel-Teensy-4.0

Hopefully this spurs some more testing and development on the SmartMatrix libraries at some point.

bleckers, 2020-03-09:
He's added the Teensy 4.0 support to octows2812!

https://github.com/PaulStoffregen/OctoWS2811/commit/35c6ff24cf0b598cfb5339aa83bf0a2d4c607f13#diff-0076f9817fc682d184d60e665989d172

embedded-creations, 2020-04-05:
https://forum.pjrc.com/threads/60337-LED-Matrix-driver-for-T4-0-using-FlexIO-parallel-out-FlexPWM-DMA-amp-SmartLED-shield

headkaze, 2020-05-14:
Teensy 4.1 has been released. Is official support more likely with this board?

embedded-creations, 2020-05-14:
Support is in progress, an update, and future updates will be posted in the linked thread:

https://community.pixelmatix.com/t/teensy-4-0-released/498/27?u=louis

embedded-creations, 2020-08-21:
For anyone not following the thread in the SmartMatrix Community, here's an announcement about SmartLED Shield for Teensy 4 and our crowdfunding campaign: https://community.pixelmatix.com/t/smartled-shield-for-teensy-4/740

embedded-creations, 2020-11-23:
Closing this as the upcoming SmartMatrix Library 4.0 has full Teensy 4 support

closed 2020-11-23 by embedded-creations

bleckers, 2020-11-25:
Fantastic effort from everyone!