pixelmatix/SmartMatrix #92: Chipset ICN2037 and RT5957

pyzimmer, 2019-11-14:
Hello, I bought this panel  : https://fr.aliexpress.com/item/32836331716.html?spm=a2g0s.9042311.0.0.232f6c37cu1lnO

The chipset is : 
- ICN2037 : http://en.chiponeic.com/content/details36_245.html
and state graphs : https://www.mikrocontroller.net/attachment/392914/ICN2037.pdf
- RT5957 : http://www.dlfushi.com/uploads/201805/5b03e7bc77f3a.pdf
- 74HC245 chip : https://assets.nexperia.com/documents/data-sheet/74HC_HCT245.pdf

I am able to display only red lines in strange order from MODE_MAP_TESTING of MultiRowRefreshMapping.ino :
[image]
vidéo : https://photos.app.goo.gl/WxDnsUxLfPkXDhWPA

I try with MODE_MAP_REVERSE_ENGINEERING, with 4 or 8 lines height, I get only red led flickering

My controller is ESP32.
I try with the library PxMatrix, I have similar issue.  

Please note that your library works fine with another model of LED panel.
Congratulation for your work !

pyzimmer, 2019-11-26:
Do you think I have any chance to get it work by creating a new PanelMappingEntry ?

embedded-creations, 2019-11-27:
No, I think the issue is something like a wiring issue, level shifters are required for your panel, or there is a chipset incompatibility.  Sorry, I don't have any clues of what to try first after watching the video.

embedded-creations, 2019-11-27:
Make sure you're not trying to use the SmartLED Shield V4 mode if you're just directly connecting pins to the Teensy (without the flip flop circuit)

pyzimmer, 2019-11-27:
Thank you. No, the SmartLED Shield V4 mode is commented.
The wiring works fine with another panel model. I try with level shifter : same result....

pyzimmer, 2019-12-12:
hello I found a similar issue with the RT5957 chip : https://community.pixelmatix.com/t/led-module-p2-128-64-1-32-driver/414/4