pixelmatix/SmartMatrix #94: Doesn't work on NodeMCE-esp32

danewber, 2019-11-20:
I'm trying to use this on a Nodemcu-esp32.  I have been very successful on an mh-et live minikit.  None of the demos will run on the nodemcu.  I have tried 2 different models (probably knockoffs but not the point).  The code stops when the "i2s_parallel_setup_without_malloc" function is called in esp32_i2s_parallel.c file.  I have tried editing the function and have found that any time the "dev" or "cfg" variables from the parameters are used, the code will stop here.  I haven't been able to find where this code is called from yet so I'm not sure what is happening.

danewber, 2019-11-20:
It is definitely stopping during the "matrix.begin()" function.

danewber, 2019-11-20:
For additional context, I'm using the "Feature Demo".  I've added output like the following:
`  // initialize the digital pin as an output.
  pinMode(ledPin, OUTPUT);

  Serial.begin(115200);

  printf("Adding background layer\n");
  matrix.addLayer(&backgroundLayer); 
  printf("Adding scrolling layer\n");
  matrix.addLayer(&scrollingLayer); 
  printf("Adding indexed layer\n");
  matrix.addLayer(&indexedLayer); 
  printf("Layers Added");
  matrix.begin();

  printf("Matrix Begun");`

The output looks like this:

> 17:03:15.140 -> Getting started
17:03:15.140 -> Adding background layer
17:03:15.140 -> Adding scrolling layer
17:03:15.188 -> Adding indexed layer
17:03:15.188 -> Layers Added

17:03:15.188 -> Starting SmartMatrix Mallocs

17:03:15.188 -> Heap/32-bit Memory Available: 373088 bytes total, 125400 bytes largest free block
17:03:15.188 -> 8-bit/DMA Memory Available  : 288336 bytes total, 125400 bytes largest free block
17:03:15.188 -> Begin background layer
17:03:15.188 -> Background layer done
17:03:15.188 -> Begin scrolling layerSmartMatrix Layers Allocated from Heap:

17:03:15.188 -> Heap/32-bit Memory Available: 359284 bytes total, 125400 bytes largest free block
17:03:15.188 -> Starting SmartMatrix DMA Mallocs

17:03:15.264 -> sizeof framestruct: 00006000

17:03:15.264 -> DMA Memory Available before ptr1 alloc: 224548 bytes total, 113792 bytes largest free block
17:03:15.264 -> matrixUpdateFrames[0] pointer: 3FFC161C

17:03:15.264 -> DMA Memory Available before ptr2 alloc: 224548 bytes total, 113792 bytes largest free block
17:03:15.264 -> matrixUpdateFrames[1] pointer: 3FFC762C

17:03:15.264 -> Frame Structs Allocated from Heap:

17:03:15.264 -> Heap/32-bit Memory Available: 309300 bytes total, 113792 bytes largest free block
17:03:15.264 -> 8-bit/DMA Memory Available  : 224548 bytes total, 113792 bytes largest free block
17:03:15.264 -> Allocating refresh buffer:

17:03:15.264 -> lsbMsbTransitionBit of 0 requires 786432 RAM, 113792 available, leaving -672640 free: 

17:03:15.264 -> lsbMsbTransitionBit of 1 requires 393216 RAM, 113792 available, leaving -279424 free: 

17:03:15.264 -> lsbMsbTransitionBit of 2 requires 196608 RAM, 113792 available, leaving -82816 free: 

17:03:15.264 -> lsbMsbTransitionBit of 3 requires 98304 RAM, 113792 available, leaving 15488 free: 

17:03:15.264 -> Raised lsbMsbTransitionBit to 3/11 to fit in RAM

17:03:15.297 -> lsbMsbTransitionBit of 3 gives 37 Hz refresh, 120 requested: 

17:03:15.297 -> lsbMsbTransitionBit of 4 gives 75 Hz refresh, 120 requested: 

17:03:15.297 -> lsbMsbTransitionBit of 5 gives 147 Hz refresh, 120 requested: 

17:03:15.297 -> Raised lsbMsbTransitionBit to 5/11 to meet minimum refresh rate

17:03:15.297 -> Descriptors for lsbMsbTransitionBit 5/11 with 16 rows require 24576 bytes of DMA RAM

17:03:15.297 -> SmartMatrix Mallocs Complete

17:03:15.297 -> Heap/32-bit Memory Available: 284692 bytes total, 113792 bytes largest free block
17:03:15.329 -> 8-bit/DMA Memory Available  : 199940 bytes total, 113792 bytes largest free block
17:03:15.329 -> Setting up parallel I2S bus at I2S1
17:03:15.329 -> I2S setup done.
17:03:15.329 -> Made it through the begin function

danewber, 2019-11-20:
The last line "Made it through the begin function" is output I added in SmartMatrixMultiplexedCalcEsp32_Impl.h at line 337

embedded-creations, 2019-11-21:
What panel configuration are you using in your sketch?  What size width and height?

closed 2020-11-23 by embedded-creations