pixelmatix/SmartMatrix #95: esp32 - wifi - 64*64 hub75e panels

Jay-esp, 2019-12-01:
Using esp32 doit board with 160mm 64*64 p2.5 panel with icn2038s drivers.
issue 1, occasionally the initialization is not correct, some parts of the display will be dim or different color, power cycling the panel solves it so it must be an initialization timing issue
issue 2 breaks the wifi library, some functions no longer work after matrix.begin
using these libs and an almost empty sketch, webserver led on led off demo
#include <WiFi.h>
#include <WiFiUdp.h>
#include <Wire.h>
#include <TimeLib.h>
#include <SmartMatrix3.h>
any clues on how to reduce the memory usage?

embedded-creations, 2019-12-03:
There's some details in the teensylc branch README on how to reduce RAM:

>Also using the new dmaRamToKeepFreeBytes parameter when calling matrix.begin() will try to keep that amount of DMA RAM free. The SD Library requires around 28000 bytes free, so you can call matrix.begin(28000) before SD.begin().

I've not seen the dim display issue, does it ever show up if you use a simple SmartMatrix Library example, or only if you add WiFi, etc, code?

Jay-esp, 2019-12-04:
Already tried the readme suggestions, the init problem also occurs without any wifi code.
I'm also running out of pins for other hardware I need to use so i'm testing a 2 esp setup, one for wifi and other stuff and one dedicated for the display.
That should resolve the wifi issue but the first test was not a success, using json data over serial port to send the commands is way to slow, working on structures over spi now.

embedded-creations, 2019-12-04:
I think that's a good way to go for your project.  I'm working on making a generic adapter to handle this, using an ESP32 to just refresh panels, and accepting APA102 formatted data over SPI from a separate micro controller.  If you want to join this thread at the SmartMatrix Community, I'm happy to let you beta test and will share more information there as the project progresses.

https://community.pixelmatix.com/t/matrix-controller/532/4

If you're happy with a 30FPS update rate (refresh rate will be decoupled and faster), you can use an ESP32 Pico board and no other circuitry (except level shifters if you need them) to receive SPI at up to 4MHz, which is fast enough to update a 64x64 panel with a new frame 30 times a second, provided there's no gaps in the data.  It's not quite ready to go but close.  I could use some motivation in the form of collaborators/testers to keep the project moving.

SebastianObi, 2020-03-10:
Hello,
i have the problem with a 63x32 pannel on ESP32 that when wifi actifity (receive/send data) the panel flickers a little bit (not extreme but you can see it). I use the ESP32 Dev Board C from AZ-Delivery.
I don't know if there is a hardware issue maybe with the voltage regulator while wifi is in use or the timing of the DMA handling.  Have someone the same issues or maybe a fix?

embedded-creations, 2020-03-10:
Is the flicker in the refreshing of the panel, e.g. will you see the flicker even if the content isn't changing?  Or is the flicker in the updating of the refresh buffer with new content?

Maybe try this pull request and pin the task to core 1? (Arduino and WiFi uses core 0 I think, or maybe try both):

https://github.com/pixelmatix/SmartMatrix/pull/103

SebastianObi, 2020-03-10:
Thanks for the fast answer :)
I have already tryed #103. The flicker is also when the content isn't changing. Are there settings to change the refreshing speed for interval/timer? I have also tryed another library for the matrix pannels [https://github.com/mrfaptastic/ESP32-RGB64x32MatrixPanel-I2S-DMA](url). But with this the flicker is a lot more on wifi actifity as with your library.

embedded-creations, 2020-03-10:
Sorry, I haven't used WiFi on the ESP32 and don't plan to.  I've heard enough about people having issues with memory allocation and now this flickering to think that it's a good idea to run WiFi and refresh the panel at the same time.  My solution is to have a second ESP32 that refreshes the matrix, and the main controller can send it pixel data over SPI.  I'm not sure what to advise you to do.  Dave from Dave's Garage seems to have wifi and refreshing working, or at least hasn't complained about flicker, maybe join his closed group and ask him?  (he's not active on the SmartMatrix Community otherwise I'd suggest you ask there and not Facebook): https://www.facebook.com/daveplgarage/

embedded-creations, 2020-03-10:
> Are there settings to change the refreshing speed

I didn't answer your direct question...  Look for the `ESP32_I2S_CLOCK_SPEED` setting in `MatrixHardware_ESP32_V0.h`

Jay-esp, 2020-03-11:
as discussed before I ended up using 2 esp modules since the wifi libraries and this one conflicted over resources resulting in weird issues
they communicate commands and parameters in json format via their serial ports, timing is not very critical for my application
wifi, ntp clock, and webserver frontend are handled by the first esp module, all smartmatrix and animations are handled by the second
remember that the panel needs plenty of current so use a good power supply for the panel separate
I did not use level shifters and this gave some erratic results, the solution was to adjust the 5v power to the panel to a little lower voltage so that the esp's 3.3V outputs high signals are within the limit of the panels input for a high signal
so far so good

closed 2020-11-23 by embedded-creations