pixelmatix/SmartMatrix #96: esp32 rgb panel scrolling not working for 4 panels

nareshmekh3, 2019-12-24:
**Devices used:**
1. Esp32
2. P6 - INDOOR - RGB LED Matrix Panel - 1/16 Scan - 32x32 Pixels - 192mm x 192mm - HUB75
3. power supply 5v 60 amps

**Pinout connection:**
R0 --->  2 
G0 --->  15
B0 --->  4
R1 --->  16 
G1 --->  27
B1 --->  17
E --->   12 
A --->   5
B --->   18
C --->   19 
D --->   21
CLK ---> 22
STD ---> 26 
OE --->  25
GND ---> gnd

Here i am using basic example FeatureDemo with 4 panels so i changed the kMatrixWidth to 128 and kMatrixHeight = 32. Remaining code i did not changed. 
The issue is scrolling text is visible in the first panel but not visible in the remaining panels.

so can any one suggest me how to over come this issue and please check the pin connection for esp32 and rgb panel.

below i am attaching the output image.

[image]

embedded-creations, 2019-12-24:
If you post issues like this over at community.pixelmatix.com you may get more responses.

I don't think it's an ESP32 pinout/wiring issue as you're getting a valid image on the first panel.  Can you show a picture of the back of the panels?

Are the panels all the same?  What happens if you swap the first panel that's working with panel 2 or 3?

nareshmekh3, 2019-12-26:
Hi, I am using same 32x32  RGB LED Matrix panel, i tried with swapping the panels but even it is giving same issues. 

below i am sharing picture of the back of the panels.
[image]

embedded-creations, 2019-12-26:
Maybe try swapping the ribbon cables around?  You know the gray cable is working, I’m not so sure about the rainbow cables. 

Also try the sketch configured for 32x32. The first panel should propagate a similar image to the second panel

nareshmekh3, 2019-12-26:
Thank you so much for your support, its working its cable issue i tried with another cable it was working fine.

[image]

closed 2020-01-01 by embedded-creations