pixelmatix/SmartMatrix #97: Noise pixel on P10 Single Color ESP32 teensylc branch

busel7, 2019-12-26:
Hi thank you for your great library,

I tried it for P10 Single Color /4 with ESP32, Wiring I used "ESP32 forum wiring", but there is alot of noise pixel, is there a solution for this?

Here is the photo : https://photos.app.goo.gl/sQcXPeNkMcT47iSj7

And here is my code : 

`#include <Wire.h>
#include <RtcDS3231.h>
RtcDS3231<TwoWire> Rtc(Wire);
//#include <SmartLEDShieldV4.h>  // uncomment this line for SmartLED Shield V4 (needs to be before #include <SmartMatrix3.h>)
#include <SmartMatrix3.h>

#define COLOR_DEPTH 24                  // known working: 24, 48 - If the sketch uses type `rgb24` directly, COLOR_DEPTH must be 24
const uint8_t kMatrixWidth = 32;        // known working: 32, 64, 96, 128
const uint8_t kMatrixHeight = 16*2;       // known working: 16, 32, 48, 64
const uint8_t kRefreshDepth = 48;       // known working: 24, 36, 48
const uint8_t kDmaBufferRows = 4;       // known working: 2-4, use 2 to save memory, more to keep from dropping frames and automatically lowering refresh rate
const uint8_t kPanelType = SMARTMATRIX_HUB12_16ROW_32COL_MOD4SCAN;   // use SMARTMATRIX_HUB75_16ROW_MOD8SCAN for common 16x32 panels
const uint8_t kMatrixOptions = (SMARTMATRIX_OPTIONS_HUB12_MODE);      // see http://docs.pixelmatix.com/SmartMatrix for options
const uint8_t kIndexedLayerOptions = (SMARTMATRIX_OPTIONS_HUB12_MODE);

SMARTMATRIX_ALLOCATE_BUFFERS(matrix, kMatrixWidth, kMatrixHeight, kRefreshDepth, kDmaBufferRows, kPanelType, kMatrixOptions);
SMARTMATRIX_ALLOCATE_INDEXED_LAYER(indexedLayer, kMatrixWidth, kMatrixHeight, COLOR_DEPTH, kIndexedLayerOptions);

const int defaultBrightness = (25*255)/100;        // full (100%) brightness
const int defaultScrollOffset = 6;
const rgb24 defaultBackgroundColor = {0, 0, 0};

RtcDateTime now;

void setup() {
  Serial.begin(115200);
  delay(200);

  // setup matrix
  matrix.addLayer(&indexedLayer); 
  matrix.begin();

  /* I2C Changes Needed for SmartMatrix Shield */
  // switch pins to use 16/17 for I2C instead of 18/19, after calling matrix.begin()//
  pinMode(18, INPUT);
  pinMode(19, INPUT);
  Wire.begin(16,17);
  Rtc.Begin();

  if (!Rtc.GetIsRunning()) {
    
    Rtc.SetIsRunning(true);
    
  }
  
  Rtc.Enable32kHzPin(false);
  Rtc.SetSquareWavePin(DS3231SquareWavePin_ModeNone); 

  matrix.setBrightness(defaultBrightness);
}

void loop() {

  RtcDateTime now = Rtc.GetDateTime();
  
  char jam[9];
  char tanggal[9];

  // clear screen before writing new text
  indexedLayer.fillScreen(0);

  /* Draw Clock to SmartMatrix */
  uint8_t hour = now.Hour();
  sprintf(jam, "%02d:%02d:%02d", hour, now.Minute(), now.Second());
  sprintf(tanggal, "%02d-%02d-%02d", now.Day(), now.Month(), now.Year()-2000);
  indexedLayer.setFont(font3x5);
  indexedLayer.drawString(0, 1,  1, jam);
  indexedLayer.drawString(0, 10, 1, tanggal);  
  indexedLayer.swapBuffers();
  delay(1000);
  
}`

Thank you in advance

embedded-creations, 2020-01-01:
There's so much noise I can't exactly tell what's going on.

- I don't see any noise on the right half of the panel, is that accurate?
- Is there any pattern to the noise?  Are the correct pixels always lit, or are they sometimes shifted?  Does the noise seem to come from valid pixels nearby vertically or horizontally?

TonivdW, 2020-01-02:
I also had a lot of noise on my panels. It went away completely after I lowered the voltage on the panels to about 4 Volts.

Seumi, 2020-01-15:
> I also had a lot of noise on my panels. It went away completely after I lowered the voltage on the panels to about 4 Volts.

How do you reduce the voltage? I also have this problem.I have tried many power adapters. The voltage is between 4.5 and 5 but the noise always exists. However, it does seem that as you said, the lower the voltage, the less noise there seems to be. At first I didn't pay attention to this phenomenon, but you reminded me, so I want to give it a try.

liebman, 2020-01-15:
Try using a level shifter or buffer like a 74LS245.  This solved my noise issues.

busel7, 2020-02-03:
> There's so much noise I can't exactly tell what's going on.
>
> * I don't see any noise on the right half of the panel, is that accurate?
> * Is there any pattern to the noise?  Are the correct pixels [...]

Hi thank your reply,
1. Yes You right, on right half panel clear, no noise
2. I'm not sure if that is a pattern noise or random noise. mostly horizontal noise few of them vertical

embedded-creations, 2020-02-06:
Seems like an issue with the panel to me.  I can't think of an issue with the library or the connection between the ESP32 and the panel that would cause noise on only horizontal half of the panel but not the other.

busel7, 2020-02-21:
> Seems like an issue with the panel to me. I can't think of an issue with the library or the connection between the ESP32 and the panel that would cause noise on only horizontal half of the panel but not the other.

Ok, thanks for your response, I will try using Buffer first.

busel7, 2020-02-21:
> Try using a level shifter or buffer like a 74LS245. This solved my noise issues.

Hi, I'm not a technical person, may I know how to wiring if I use 74HC245D, from ESP32 > Buffer > P10 HUB12. 

I read the datasheet but still confuse how to use it. Please help :)

busel7, 2020-03-22:
> Seems like an issue with the panel to me. I can't think of an issue with the library or the connection between the ESP32 and the panel that would cause noise on only horizontal half of the panel but not the other.

Hi, thanks for update. Just try the latest update, it looks all good, but.

1. P10-HUB12-DIP-LED-Blue looks clean
[image]

2. P10-HUB12-DIP-LED-Green looks clean with little shadow
[image]

3. P10-HUB12-SMD-LED-Green looks clean, the shadow is clearly visible
[image]

closed 2020-03-22 by busel7

busel7, 2020-04-11:
For the shaded problem I overcome it by using two different power supplies, one for the controller, one for the led panel. This is a great library for led panels. Thank you very much Mr. Louis Beaudoin

luceec, 2020-10-22:
Hi there, can someone post the link to the ESP32 HUB12 wiring please?

busel7, 2020-10-22:
> Hi there, can someone post the link to the ESP32 HUB12 wiring please?

Hi, I noted on my video desciption here https://youtu.be/YkShhMBcBQw

[image]

luceec, 2020-10-22:
Thank you man!

32byteElectronics, 2021-02-02:
[image]
In this Image C is not present @busel7

busel7, 2021-02-03:
> [image]
> In this Image C is not present @busel7
> @busel7 I want to connect esp32 with this p10 module.

you can ignore a C pin (not used)

FARRAZ707, 2025-11-21:
> I also had a lot of noise on my panels. It went away completely after I lowered the voltage on the panels to about 4 Volts.

does it make the brightness lowered to