pixelmatix/SmartMatrix #98: 'BUILTIN_SDCARD' was not declared in this scope

pinkies-jewel, 2019-12-31:
Hello. I have used this library before with a 32x32 as well as a teensy 3.6. I am now trying with a different teensy 3.6 and a 64x64 led shield. I am having the issue of a 'BUILTIN_SDCARD' was not declared in this scope. I am not sure how to fix this. I am attaching a screenshot for reference. I haven't done anything to the code except change the definition to include the builtin_sdcard as well as change it to be the 64x64 panel. Help is appreciated in advance :)
[image]

pinkies-jewel, 2019-12-31:
Here is the code for reference in case I did something wrong accidentally.
[AnimatedGIFs.txt](https://github.com/pixelmatix/SmartMatrix/files/4012058/AnimatedGIFs.txt)

embedded-creations, 2020-01-01:
can you turn on verbose output for compilation in settings, compile, and paste what you see?

Did you select the Teensy 3.6 board type?

closed 2020-11-23 by embedded-creations