pixelmatix/SmartMatrix #100: pixel columns off by 2 (ESP32 teensylc branch)

TonivdW, 2020-01-02:
I have the demos in the teensylc branch running on an ESP32 on a 64x64 panel. 
Problem is that the image is shifted two pixel columns to the left, with the two leftmost columns being displayed on the far right side. Any idea what might be wrong?

embedded-creations, 2020-01-02:
What’s your hardware setup?

TonivdW, 2020-01-02:
Thanks for the fast response! I'm however not exactly sure what you are asking.
I have wired the ESP32 to the HUB75 of the panel as described within ESP32_FORUM_PINOUT in MatrixHardware_ESP32_V0.h
kPanelType = SMARTMATRIX_HUB75_64ROW_MOD32SCAN
I have also tried linking two 64x64 panels creating an 128x64 display and this has the same issue.
What more info can I give you?

embedded-creations, 2020-01-02:
That's useful.  I assume no level shifters?

Try reversing the changes in this commit.  I didn't do enough testing on this change and it may have broken something:

https://github.com/pixelmatix/SmartMatrix/commit/ebf12e4fa3d212ded543970450ad4bc1742ad3fd

TonivdW, 2020-01-02:
I reversed the change and now just one column is misplaced. The leftmost column is displayed in the rightmost position.

embedded-creations, 2020-01-05:
Sorry, I'm not sure what else could be going on, and I don't have time to try to reproduce it here right now

KiloBGram, 2020-04-25:
I have exactly the same issue. The leftmost column is on the rightmost edge. Tried messing around with options to see if anything would fix it, but no.

embedded-creations, 2020-04-25:
Try lowering the CLK speed.  Default is 20 MHz.  Try as low as 1 MHz, see what happens

in `MatrixHardware_ESP32_V0.h`

`#define ESP32_I2S_CLOCK_SPEED (20000000UL)`

fcapano, 2020-05-03:
One thing I noticed on my panel is that this happens depending on which pin I choose as CLK. E.g. with the default CLK pin (16) my panel works fine. If I switch CLK to pin 22, the leftmost column is drawn on the right.

closed 2020-11-23 by embedded-creations