pixelmatix/SmartMatrix #104: Can not work with 32*32 1/8 scan led panel(teensylc)

Seumi, 2020-01-13:
I got a 32*32 1/8 scan P5 panel which look this
[image]
In theory, 1/8 scan will light twice as many leds as 1/16(which is more common for a 32 rows panel) at the same time, so it will have higher brightness.
But there are no define for this type panel in SmartMatrixMultiplexedCommon.h which should like
`#define SMARTMATRIX_HUB75_32ROW_MOD8SCAN`. I just use 'SMARTMATRIX_HUB75_32ROW_MOD16SCAN', of course it does not work normally.
[image]
(use Bitmaps in example, should be a chrome logo)
So, i add a define 'SMARTMATRIX_HUB75_32ROW_MOD8SCAN' and  modified the code like this
`#define CONVERT_PANELTYPE_TO_MATRIXPANELHEIGHT(x)   ((x == SMARTMATRIX_HUB75_32ROW_MOD8SCAN ? 32 : 0))
#define CONVERT_PANELTYPE_TO_MATRIXROWPAIROFFSET(x) ((x == SMARTMATRIX_HUB75_32ROW_MOD8SCAN ? 8 : 0))
#define CONVERT_PANELTYPE_TO_MATRIXSCANMOD(x) ((x == SMARTMATRIX_HUB75_32ROW_MOD8SCAN ? 8 : 0))
`
Unfortunately, it did not work as expected and even more messy than 16scan.
[image]
After google , i found [Everything You Didn't Want to Know About RGB Matrix Panels (When SparkFun got a shipment of 1:8 scan rate panels, it fell on me to make them work with the wrong library.)](https://www.sparkfun.com/sparkx/blog/2650), it says that the 1/8 scan rate has a 'Weird Scan Order' compared with 1/16 scan with 32*32 panel . It fill up the first 16 slots of the first row, then it jumps to the eighth row and fills all 32 slots before jumping back up to the first for the last 16 before moving on to the second row.
Therefore, I believe there are other code of the smartmatrix should be modified. Who has encountered the same problem, and are there any solutions or suggestions? 
Thanks.

Seumi, 2020-01-13:
I find 'SmartMatrixPanelMaps.cpp' may has something to do with this problem. I add this code
`const PanelMappingEntry panelMap32x32Mod8[] =
{
    {0, 0,  8},
    {0, 8,  8},
    {0, 48, 8},
    {0, 56, 8},
    {8, 16, 8},
    {8, 24, 8},
    {8, 32, 8},
    {8, 40, 8}, 
    {0, 0, 0}   // last entry is all zeros
};`
follow the 'Weird Scan Order'. Still not work :<

Seumi, 2020-01-14:
`
const PanelMappingEntry panelMap32x32Mod8[] =
{
    {0, 0, 32},
    {8, 0, 32},
    {0, 0, 0} // last entry is all zeros
};
`
It worked with this, but i have another problem, maybe a new issue, so close this

closed 2020-01-14 by Seumi

whiteshavedchocolate, 2020-03-26:
Hi, Seumi. Thank you for your determination with modifying the code, so that you can display on a 32*32 1/8 scan LED panel.

I have a similar situation: I have a 64*64 1/16 scan LED panel. I also can get something to display, only if I use SMARTMATRIX_HUB75_32ROW_MOD16SCAN. However, as you would guess, it isn't displaying properly. 

Based on what you learned, can you advise what I need to do to get my panel to display correctly? Do you know how my const PanelMappingEntry panelMap64x64Mod16[] would need to be defined? Thank you!