pixelmatix/SmartMatrix #109: [Question] teensylc: Possible to use I2S microphone in combination with RGB Panel?

HorstBaerbel, 2020-02-19:
I have a 32x16 1/8 scan panel working here nicely with a LOLIN32 and the regular "Forum pinout". I have a microphone breakout attached to the ADC at A0 / VP (512 samples, 9 bit, 0dB, full 0-3.3V range), but it is very noisy, non-linear and sampling takes quite some time (~8-9 ms).
-> I would like to use a I2S MEMS microphone like the SPH 0645 or better the INMP441 or ICS 42432 / 43434. Can I still connect (and properly configure) the microphone to the other I2S ports (and what would be the pins) or would that interfere with SmartMatrix?
-> If not, are there alternatives to improve microphone input?

embedded-creations, 2020-02-19:
There are two I2S peripherals, and SmartMatrix Library uses I2S1, but with some additional code added could use I2S0 if required.  I don't know how to hook up an I2S microphone, but I'm guessing it can be hooked up to most any pins if the I2S Microphone library allows for selecting pins.

If you get it to work, could you please post back?  Also LMK if you have more questions.

embedded-creations, 2020-02-19:
Dave has SmartMatrix Library working with an I2S microphone if I remember correctly.  You might try contacting him: https://www.facebook.com/daveplgarage/ (more details available in his closed Dave's Garage Group, which you may be able to join on request)

HorstBaerbel, 2020-02-19:
Thanks for answering. I have no Facebook though. Good to know SmartMatrix is using only one I2S peripheral. It seems I2S0 can be routed almost anywhere through the GPIO matrix, so I'll try to get a microphone and connect it.

HorstBaerbel, 2020-06-16:
I used [this project](https://github.com/ikostoski/esp32-i2s-slm) with an INMP441 and made it work with SmartMatrix. Will post again when I have commited the code to GitHub...

embedded-creations, 2020-11-23:
I'm going to close this, but if you ever add your code to GitHub please LMK

closed 2020-11-23 by embedded-creations

HorstBaerbel, 2021-07-19:
I've uploaded the code here now. There's still some bits and pieces missing missing. Will update in the next days.