pixelmatix/SmartMatrix #110: Adapt smartmatrix to Espruino

B1tb0t, 2020-03-09:
Hi, I realy love the smartmatrix library for my Esp32. My latest project is based on Espruino, but I want to use smartmatrix with it (there is no libaray for Hub75 led matrix yet). Is it possible to adapt this library to Espruino?

Jonas

embedded-creations, 2020-03-10:
I’m not sure how that would work, and it’s not a trivial project. Is there an APA102 LED library for Esprino?

B1tb0t, 2020-03-10:
I found no full libarary for APA102. There is some code to controll APA102 led strips with SPI.
I am currently trying to include smartmatrix in the espruino firmware (https://www.espruino.com/Extending+Espruino+1). But it needs the arduino.h file to work, but I found no way to include it yet.

embedded-creations, 2020-03-10:
Maybe instead of including Arduino.h, remove the #include and see what definitions are missing and add them directly instead. It might just be a few definitions I use from Arduino, I’m not sure

B1tb0t, 2020-03-10:
Ok I removed the Arduino.h including and build again. Now there are a lot of small errors:

`In file included from libs/smartmatrix/src/smartmatrix/SmartMatrix3.h:40:0,
                 from libs/smartmatrix/jswrap_smartmatrix.c:20:
libs/smartmatrix/src/smartmatrix/MatrixHardware_ESP32_V0.h:135:13: note: #pragma message: ESP32 forum wiring
     #pragma message "ESP32 forum wiring"
             ^
libs/network/esp32/jswrap_esp32_network.c:1555:82: warning: conversion to 'JsVarInt {aka int}' from 'uint32_t {aka unsigned int}' may change the sign of the result [-Wsign-conversion]
     jsvObjectSetChildAndUnLock(jsPingResponse, "totalBytes",   jsvNewFromInteger(pingResp->total_byte                                                                                  ^
In file included from libs/smartmatrix/src/smartmatrix/SmartMatrix3.h:43:0,
                 from libs/smartmatrix/jswrap_smartmatrix.c:20:
libs/smartmatrix/src/smartmatrix/MatrixCommon.h:40:5: error: expected specifier-qualifier-list before 'rgb24'
     rgb24() : rgb24(0,0,0) {}
     ^
libs/network/esp32/jswrap_esp32_network.c:1556:82: warning: conversion to 'JsVarInt {aka int}' from 'uint32_t {aka unsigned int}' may change the sign of the result [-Wsign-conversion]
     jsvObjectSetChildAndUnLock(jsPingResponse, "totalTime",    jsvNewFromInteger(pingResp->total_time                                                                                  ^
libs/smartmatrix/src/smartmatrix/MatrixCommon.h:53:5: error: expected specifier-qualifier-list before 'rgb48'
     rgb48() : rgb48(0,0,0) {}
     ^
libs/network/esp32/jswrap_esp32_network.c:1557:82: warning: conversion to 'JsVarInt {aka int}' from 'uint32_t {aka unsigned int}' may change the sign of the result [-Wsign-conversion]
     jsvObjectSetChildAndUnLock(jsPingResponse, "respTime",     jsvNewFromInteger(pingResp->resp_time)                                                                                  ^
In file included from libs/smartmatrix/src/smartmatrix/SmartMatrix3.h:43:0,
                 from libs/smartmatrix/jswrap_smartmatrix.c:20:
libs/smartmatrix/src/smartmatrix/MatrixCommon.h:65:13: error: expected identifier or '(' before '&' token
 inline rgb24& rgb24::operator=(const rgb24& col) {
             ^
libs/network/esp32/jswrap_esp32_network.c:1560:82: warning: conversion to 'JsVarInt {aka int}' from 'uint32_t {aka unsigned int}' may change the sign of the result [-Wsign-conversion]
     jsvObjectSetChildAndUnLock(jsPingResponse, "timeoutCount", jsvNewFromInteger(pingResp->timeout_co                                                                                  ^
libs/smartmatrix/src/smartmatrix/MatrixCommon.h:72:13: error: expected identifier or '(' before '&' token
 inline rgb24& rgb24::operator=(const rgb48& col) {
             ^
libs/network/esp32/jswrap_esp32_network.c:1561:82: warning: conversion to 'JsVarInt {aka int}' from 'uint32_t {aka unsigned int}' may change the sign of the result [-Wsign-conversion]
     jsvObjectSetChildAndUnLock(jsPingResponse, "bytes",        jsvNewFromInteger(pingResp->bytes));                                                                                    ^
libs/smartmatrix/src/smartmatrix/MatrixCommon.h:79:13: error: expected identifier or '(' before '&' token
 inline rgb48& rgb48::operator=(const rgb24& col) {
             ^
libs/network/esp32/jswrap_esp32_network.c:1562:82: warning: conversion to 'JsVarInt {aka int}' from 'uint32_t {aka unsigned int}' may change the sign of the result [-Wsign-conversion]
     jsvObjectSetChildAndUnLock(jsPingResponse, "error",        jsvNewFromInteger(pingResp->err_count)                                                                                  ^
libs/network/esp32/jswrap_esp32_network.c:1548:40: warning: unused parameter 'msgType' [-Wunused-parameter]
 esp_err_t pingResults(ping_target_id_t msgType, esp_ping_found * pingResp){
                                        ^
libs/smartmatrix/src/smartmatrix/MatrixCommon.h: In function 'calculateBackgroundLUT':
libs/network/esp32/jswrap_esp32_network.c: In function 'jswrap_wifi_ping':
libs/smartmatrix/src/smartmatrix/MatrixCommon.h:203:18: warning: conversion to 'uint16_t {aka short unsigned int}' from 'int' may alter its value [-Wconversion]
         lut[i] = (lightPowerMap16bit[i] * backgroundBrightness) / 256;
                  ^
libs/network/esp32/jswrap_esp32_network.c:1580:15: warning: conversion to 'int' from 'size_t {aka unsigned int}' may change the sign of the result [-Wsign-conversion]
     int len = jsvGetString(ipAddr, ipString, sizeof(ipString)-1);
               ^
libs/smartmatrix/src/smartmatrix/MatrixCommon.h: At top level:
libs/network/esp32/jswrap_esp32_network.c:1590:15: warning: conversion to 'u32_t {aka unsigned int}' from 'JsVarInt {aka int}' may change the sign of the result [-Wsign-conversion]
     ip.addr = jsvGetInteger(ipAddr);
               ^
libs/smartmatrix/src/smartmatrix/MatrixCommon.h:212:10: error: expected '=', ',', ';', 'asm' or '__attribute__' before '<' token
 template <typename RGB_IN>
          ^
libs/network/esp32/jswrap_esp32_network.c:1625:63: warning: invalid application of 'sizeof' to a function type [-Wpointer-arith]
   esp_ping_set_target(PING_TARGET_RES_FN, &pingResults, sizeof(pingResults));
                                                               ^
libs/smartmatrix/src/smartmatrix/MatrixCommon.h:219:10: error: expected '=', ',', ';', 'asm' or '__attribute__' before '<' token
[... cut ...]
                     ^
src/jsvar.c:3903:13: warning: conversion to 'size_t {aka unsigned int}' from 'int' may change the sign of the result [-Wsign-conversion]
           i += jsvGetFlatStringBlocks(v); // skip forward
             ^
src/jsvar.c:3903:16: warning: conversion to 'int' from 'size_t {aka unsigned int}' may change the sign of the result [-Wsign-conversion]
           i += jsvGetFlatStringBlocks(v); // skip forward
                ^
make: *** wait: No child processes.  Stop.`

Is it a version problem ?

embedded-creations, 2020-03-10:
Arduino.h probably includes a other required header files, like inttypes.h (not sure if that's the actual name).  I can't tell you exactly how to fix this problem, but I'd start by finding Arduino.h in an Arduino install and adding it to your project, or including what's necessary from it.

B1tb0t, 2020-03-10:
I dont know mutch about led matrixes, but is it possible to write a matrix driver for the espruino graphics library (https://www.espruino.com/Graphics) ?

embedded-creations, 2020-03-25:
It might be possible, but not without a lot of work, which I'm not going to be doing myself.  I'm going to close this issue.

closed 2020-03-25 by embedded-creations