pixelmatix/SmartMatrix #113: DMAChannel.h is required when using for ESP32

mcsarge, 2020-03-30:
In trying to use this library, it requires DMAChannel.h from Teensy. I am using this for ESP32 development in Platformio:
platform = espressif32
board = esp-wrover-kit
framework = arduino

So I am confused. Is there some other setup I need to do to get this to compile the examples?

mcsarge, 2020-03-30:
Additionally, I fired up the Arduino environment and tried to do the same and got the same problem. I am not using a Teensy, but an ESP32.

embedded-creations, 2020-03-30:
Are you using the Teensylc branch?

mcsarge, 2020-03-31:
I beleive so.

mcsarge, 2020-03-31:
Pilot error, please close this, evidently if you install a library from the disk, it caches some of it. I needed to set the branch first and do a pull, then install it into platformio. So sorry for the interruption.

closed 2020-03-31 by mcsarge