pixelmatix/SmartMatrix #115: Please provide examples in the Platformio environment.

mcsarge, 2020-04-24:
Use of the library would be greatly enhanced if there were clear example of how to use it within the Platformio environment. Particularly when you are trying to get it to work with the ESP32. It can be a bit tricky to clone the repository, change the branch to "teensylc" then use the direct "Add Library" to get the correct version of the library to work with and ESP32.

embedded-creations, 2020-05-07:
I don't use Platformio.  If you want to contribute the changes, or some instructions that would help others that use Platformio, I'm open to adding them to the repository

mcsarge, 2020-05-14:
I can do that, but I am unsure if I am doing it the proper way - I use git to get the repository, switch to the teensylc branch. Then I use platformio to add a local library - pointing at the library on my computer. Platformio has an awsome library manager and someone has added a library called  "SmartMatrix GFX" but it is not your library.

If this sounds like the way to do it, I will write out the instructions and compiling one of the examples.

mcsarge, 2020-05-14:
I will make it a post on my Blog, but we should be able to lift it and put it here on the Git, I will let you tell me where to put it.

embedded-creations, 2020-05-20:
Thanks for the heads up @mcsarge , I wasn't aware of the Library Manager.  It looks like the released version of SmartMatrix Library (the one without ESP32 support) can be added to the Library Manager, as it recognizes the Arduino Library Manager format.  That doesn't seem useful for you though.  I'm planning a SmartMatrix Library 4.0 release which will bring the ESP32 and other changes out of the teensylc branch into a new repository, which can then be added to both Arduino and PlatformIO Library Managers.

https://docs.platformio.org/en/latest/librarymanager/creating.html

embedded-creations, 2020-05-20:
Oops, didn't finish my thought... In the meantime, adding SmartMatrix Library as a local library seems like the way to go, I think you're on the right track

mcsarge, 2020-05-27:
I have made a bunch of screen shots of using a local library and programming an ESP32 and running the FeatureDemo which I will put together in a post and we can go from there. Just a lot on my plate at the moment, but I will get to it.

embedded-creations, 2020-05-27:
Thanks for the update, no rush!

embedded-creations, 2020-11-23:
Hi @mcsarge, I'm going to close this, but if you have any documentation to share, please let me know

mcsarge, 2020-11-23:
OK, I am sorry I got distracted on this subject. Did you create the new version yet that moves the ESP32 out of the Teensy branch? I hope to get back to this in the dark of Winter :-)

embedded-creations, 2020-11-23:
No problem.  SmartMatrix Library 4.0 will be released shortly, and it will have all the ESP32 support in the main branch, and released through Arduino Library Manager if that's what you're asking.  I'll open this back up.  Also, I'm planning of making the SmartMatrix Library documentation into an open wiki, so you can just add info to the wiki if you want.

embedded-creations, 2021-02-08:
to @mcsarge or anyone who may have registered the "pixelmatix" username in PlatformIO and registered SmartMatrix Library and GifDecoder:

I'm starting to use PlatformIO now, and want to keep these libraries up to date with PlatformIO.  Someone has already created a username for "pixelmatix" and registered SmartMatrix Library and GifDecoder, which is great except that I can't update these library registry entries as new releases are made.  The registries are tied to the "pixelmatix" username, which I don't have.  I can create a new username, e.g. "embedded-creations", and register the updated libraries, but the "pixelmatix" registries are still there and take priority, and it creates a mess if there's two registries with the same name for each library.

If you have the "pixelmatix" username, can you please get in touch so we can figure this out?

embedded-creations, 2021-02-08:
Got it figured out, PlatformIO granted me access to the "pixelmatix" organization and I can now submit updates to SmartMatrix Library and GifDecoder

mcsarge, 2021-02-09:
Cool, because it was not me! Looking forward to your work!

Matt