pixelmatix/SmartMatrix #116: Inverted I2S CLK needed

fcapano, 2020-04-25:
Commit fcca3a2906395612038c7664b857afcce8f5ce13 broke my RGB panel.

I don't know the rationale behind adding/reverting the inversion of the I2S CLK, but perhaps this could be toggled on/off with a SMARTMATRIX_ALLOCATE_BUFFERS(..., kMatrixOptions) flag?

I've attached a video with latest head before and after reverting the commit

[image]

[image]

.

embedded-creations, 2020-04-25:
Sure, can you give this a try?  https://github.com/pixelmatix/SmartMatrix/commit/42e9756038c6ad4e70d51c385099260a6f988a9d

fcapano, 2020-04-25:
Well that was fast!
It's working now, thanks :)

closed 2020-04-25 by fcapano