pixelmatix/SmartMatrix #117: P5 32x64 not correct display

gunaone, 2020-05-01:
i  was try with this p5 panel 32x64 but i got looklike this,
can anyone help,

embedded-creations, 2020-05-01:
what settings are you using at the top of your sketch?  can you post a photo of the back of the panel?  can you follow the instructions and run the multirowmapping sketch, and post a video of the sketch running from the beginning?

gunaone, 2020-05-02:
i use example futuredemo 
with seting :
````
#define COLOR_DEPTH 24                  // known working: 24, 48 - If the sketch uses type `rgb24` directly, COLOR_DEPTH must be 24
const uint8_t kMatrixWidth = 64;        // known working: 32, 64, 96, 128
const uint8_t kMatrixHeight = 32;       // known working: 16, 32, 48, 64
const uint8_t kRefreshDepth = 36;       // known working: 24, 36, 48
const uint8_t kDmaBufferRows = 4;       // known working: 2-4, use 2 to save memory, more to keep from dropping frames and automatically lowering refresh rate
const uint8_t kPanelType = SMARTMATRIX_HUB75_32ROW_MOD16SCAN;   // use SMARTMATRIX_HUB75_16ROW_MOD8SCAN for common 16x32 panels
const uint8_t kMatrixOptions = (SMARTMATRIX_OPTIONS_NONE);      // see http://docs.pixelmatix.com/SmartMatrix for options
const uint8_t kBackgroundLayerOptions = (SM_BACKGROUND_OPTIONS_NONE);
const uint8_t kScrollingLayerOptions = (SM_SCROLLING_OPTIONS_NONE);
const uint8_t kIndexedLayerOptions = (SM_INDEXED_OPTIONS_NONE);

````

.

embedded-creations, 2020-05-07:
> follow the instructions and run the multirowmapping sketch, and post a video of the sketch running from the beginning

ondoteam, 2020-05-11:
Hello.

I am in troubles with a 64x32 panel too. It was working with PxMatrix. I am using a ESP32 wired directly to the panel.

This is my current config for MultiRowRefreshMapping:

```
#define GPIOPINOUT ESP32_FORUM_PINOUT_WITH_LATCH
#define MODE_MAP_REVERSE_ENGINEERING  0
#define MODE_MAP_TESTING              1

//#define SKETCH_MODE   MODE_MAP_REVERSE_ENGINEERING
#define SKETCH_MODE   MODE_MAP_TESTING

#define COLOR_DEPTH 24                  // known working: 24, 48 - If the sketch uses type `rgb24` directly, COLOR_DEPTH must be 24
const uint8_t kMatrixWidth = 64;        // known working: 16, 32, 48, 64
const uint8_t kMatrixHeight = 32;       // known working: 32, 64, 96, 128
const uint8_t kPanelType = SMARTMATRIX_HUB75_32ROW_MOD16SCAN;   // use SMARTMATRIX_HUB75_16ROW_MOD8SCAN for common 16x32 panels
const uint8_t kRefreshDepth = 36;       // known working: 24, 36, 48
const uint8_t kDmaBufferRows = 4;       // known working: 2-4, use 2 to save memory, more to keep from dropping frames and automatically lowering refresh rate
const uint8_t kMatrixOptions = (SMARTMATRIX_OPTIONS_NONE);      // see http://docs.pixelmatix.com/SmartMatrix for options
const uint8_t kBackgroundLayerOptions = (SM_BACKGROUND_OPTIONS_NONE);
```

It is what you need? Sorry for my bad english.

embedded-creations, 2020-05-11:
@ondoteam What is not working?  Can you provide a video of the sketch running in MODE_MAP_REVERSE_ENGINEERING mode, and in MODE_MAP_TESTING mode?

ondoteam, 2020-05-11:
Sure. With the default values?

```
#if (SKETCH_MODE == MODE_MAP_REVERSE_ENGINEERING)
const uint8_t kMatrixWidth = 128;        // known working: 16, 32, 48, 64
const uint8_t kMatrixHeight = 4;       // known working: 32, 64, 96, 128
const uint8_t kPanelType = SMARTMATRIX_HUB75_4ROW_MOD2SCAN;   // Use this to reverse engineer mapping for a MOD2 panel
//const uint8_t kPanelType = SMARTMATRIX_HUB75_8ROW_MOD4SCAN;   // Use this to reverse engineer mapping for a MOD4 panel
#endif
```

embedded-creations, 2020-05-11:
@ondoteam No, follow the instructions in the Sketch and pick the values that make sense for your panel.

I haven't seen any 128x64 panels that need this mapping feature, so I'm not sure that's what's causing the issue.  You may just want to run something simple like the Bitmaps sketch and take a video of that.

```
const uint8_t kMatrixWidth = 128;        // known working: 16, 32, 48, 64
const uint8_t kMatrixHeight = 64;       // known working: 32, 64, 96, 128
const uint8_t kPanelType = SMARTMATRIX_HUB75_64ROW_MOD32SCAN;
```
These values are what I would try first for your panel with the Bitmaps sketch

ondoteam, 2020-05-11:
It is not a 128x64. It is 64x32 16SCAN

[image]
[image]
[image]

ondoteam, 2020-05-11:
This is what I am getting for the following config in the bitmap example:

```
#include <SmartMatrix3.h>
#define GPIOPINOUT ESP32_FORUM_PINOUT_WITH_LATCH
#include "gimpbitmap.h"

// pixelmatix.c is a bitmap exported from GIMP without modification
#include "pixelmatix.c"

// colorwheel.c has been modified to use the gimp32x32bitmap struct
#include "colorwheel.c"

// chrome16 is a 16x16 pixel bitmap, exported from GIMP without modification
#include "chrome16.c"

#define COLOR_DEPTH 24                  // known working: 24, 48 - If the sketch uses type `rgb24` directly, COLOR_DEPTH must be 24
const uint8_t kMatrixWidth = 64;        // known working: 16, 32, 48, 64
const uint8_t kMatrixHeight = 32;       // known working: 32, 64, 96, 128
const uint8_t kRefreshDepth = 36;       // known working: 24, 36, 48
const uint8_t kDmaBufferRows = 4;       // known working: 2-4, use 2 to save memory, more to keep from dropping frames and automatically lowering refresh rate
const uint8_t kPanelType = SMARTMATRIX_HUB75_32ROW_MOD16SCAN; // use SMARTMATRIX_HUB75_16ROW_MOD8SCAN for common 16x32 panels, or use SMARTMATRIX_HUB75_64ROW_MOD32SCAN for common 64x64 panels
const uint8_t kMatrixOptions = (SMARTMATRIX_OPTIONS_NONE);      // see http://docs.pixelmatix.com/SmartMatrix for options
const uint8_t kBackgroundLayerOptions = (SM_BACKGROUND_OPTIONS_NONE);
```

[image]
[image]

ondoteam, 2020-05-11:
There is a bad contact in the panel on OE (maybe this afternoon I forced the connector). I will check it tomorrow and update this issue. Thank you very much @embedded-creations

ondoteam, 2020-05-14:
Hello. It is working now. Was a weak connection in the PCB. Finally I am using this config:

```
#include <SmartMatrix3.h>
#define GPIOPINOUT ESP32_FORUM_PINOUT_WITH_LATCH
#define COLOR_DEPTH 24                  // known working: 24, 48 - If the sketch uses type `rgb24` directly, COLOR_DEPTH must be 24
const uint8_t kMatrixWidth = 64;        // known working: 32, 64, 96, 128
const uint8_t kMatrixHeight = 32;       // known working: 16, 32, 48, 64
const uint8_t kRefreshDepth = 36;       // known working: 24, 36, 48
const uint8_t kDmaBufferRows = 4;       // known working: 2-4, use 2 to save memory, more to keep from dropping frames and automatically lowering refresh rate
const uint8_t kPanelType = SMARTMATRIX_HUB75_32ROW_MOD16SCAN; // use SMARTMATRIX_HUB75_16ROW_MOD8SCAN for common 16x32 panels, or use SMARTMATRIX_HUB75_64ROW_MOD32SCAN for common 64x64 panels
const uint8_t kMatrixOptions = (SMARTMATRIX_OPTIONS_NONE);      // see http://docs.pixelmatix.com/SmartMatrix for options
const uint8_t kBackgroundLayerOptions = (SM_BACKGROUND_OPTIONS_NONE);
const uint8_t kScrollingLayerOptions = (SM_SCROLLING_OPTIONS_NONE);
const uint8_t kIndexedLayerOptions = (SM_INDEXED_OPTIONS_NONE);

SMARTMATRIX_ALLOCATE_BUFFERS(matrix, kMatrixWidth, kMatrixHeight, kRefreshDepth, kDmaBufferRows, kPanelType, kMatrixOptions);
SMARTMATRIX_ALLOCATE_BACKGROUND_LAYER(backgroundLayer, kMatrixWidth, kMatrixHeight, COLOR_DEPTH, kBackgroundLayerOptions);
SMARTMATRIX_ALLOCATE_SCROLLING_LAYER(scrollingLayer, kMatrixWidth, kMatrixHeight, COLOR_DEPTH, kScrollingLayerOptions);
SMARTMATRIX_ALLOCATE_INDEXED_LAYER(indexedLayer, kMatrixWidth, kMatrixHeight, COLOR_DEPTH, kIndexedLayerOptions);

```

One more time, thank you very much @embedded-creations

embedded-creations, 2020-05-14:
@ondoteam glad you got it working

@gunaone to help you I need to see a video with the results of the MultiRowMapping sketch.

gunaone, 2020-05-22:
witchone..?
i cant find "MultiRowMapping"
https://github.com/pixelmatix/SmartMatrix/tree/master/examples

embedded-creations, 2020-05-22:
@gunaone , sorry, it’s not available in the main repo, it’s on the “teensylc” branch, where you can test code that will be in the next SmartMatrix Library release

closed 2020-11-23 by embedded-creations

erickimm13, 2021-10-15:
I have 32/16 with 1/2 scan rate. that when I run the MODE_MAP_REVERSE_ENGINEERING sketch 2 red dots and 2 purple come on.

[image]

erickimm13, 2021-10-15:
the yellow and white are showing 1 dot. I have different S2 that work fine with that sketch. but not this ones. see video.

erickimm13, 2021-10-15:
[image]
this is the back of the panel that shows 2 dots.

erickimm13, 2021-10-15:
video of the panel that s2 that works with the same sketch

https://1drv.ms/v/s!Ao8XNpgZj8tdhp0vt3Udan0NviMmzw?e=f3Ku1w

embedded-creations, 2021-10-16:
Hi, you're commenting on a closed issue.  This is a better topic to discuss in a new post in community.pixelmatix.com where you may be able to find someone else who has the same issue and maybe a fix.  I don't have any suggestions at the moment.,