pixelmatix/SmartMatrix #120: [Question] level shifter problem

qlalfdu, 2020-05-19:
I am using esp32.
Recently, a level shifter (74HC244) was added and a problem occurred.
The pixel is pushed one space to the left.
Where should the library be modified to fix this?

embedded-creations, 2020-05-19:
Try the flag that inverts clock. Search the repo for it, I’m on a phone without easy access right now

qlalfdu, 2020-05-19:
Thank you Louis!
https://github.com/pixelmatix/SmartMatrix/commit/42e9756038c6ad4e70d51c385099260a6f988a9d
I did it according to the above post, but it does not change.
The newly updated library today is not pushed and is normal.
However, I am having trouble using this new library.
Because I use the timer interrupt to swap every 0.5 seconds,
Reset when swapping

embedded-creations, 2020-05-19:
> However, I am having trouble using this new library.

Can you find out what recent change started introducing resets when swapping?

> Because I use the timer interrupt to swap every 0.5 seconds,

This doesn't sound like a good idea, as there's code in swapBuffers() that can take a long time to complete.  At a minimum, use swapbuffers(false) which won't wait until the swap completes before returning (but will also not fill the new drawing buffer with the latest image.  It would be better if you created a new task to do the swap; don't do it from an interrupt.

qlalfdu, 2020-05-20:
Thank you Louis!

When swapping, it appears to be a conflict with the timer interrupt.

I have to blink every 0.5 seconds, so I have no other idea if I have to do it within the interrupt.

For example, when swapping in a loop, the screen freezes for a few seconds when checking a telegram message.

Using 'swapBuffers (false)' doesn't reset.

Thank you very much!

closed 2020-05-20 by qlalfdu