pixelmatix/SmartMatrix #121: Linker error "undefined reference to `vtable for SM Layer'"

Jkallus, 2020-05-19:
I was trying to build the examples in the teensylc branch for ESP32 and I got this error from the linker. I was using VisualGDB in Visual Studio when this happened and it doesn't seem to occur when using the Arduino IDE so its possibly an issue with VisualGDB's configuration. If it's not then perhaps someone here knows why this might be happening.

embedded-creations, 2020-05-19:
It's possible it's because I didn't create `getRequestedBrightnessShifts` functions in all the classes based on SM_Layer.  Try adding that to the other classes, you can just have it return 0.

https://github.com/pixelmatix/SmartMatrix/commit/62a2c96fb492356a2a345d941af665a9accc5769

https://github.com/pixelmatix/SmartMatrix/commit/938df2bc3fa8294392453620212d5f8cc26fbd1f

Or maybe it's something to do with another virtual method in SM_Layer

Jkallus, 2020-05-19:
I can try but I've only recently come across the library so I'm not very familiar with it. Thanks for the fast response!

Jkallus, 2020-05-20:
Never mind, I resolved it by changing some settings in VisualGDB which shouldn't have affected this.

closed 2020-05-20 by Jkallus