pixelmatix/SmartMatrix #122: Display artifacts when using generic ESP32

Jkallus, 2020-05-22:
I've been doing a test where I turn all the leds to green one by one on a 16x32 matrix and have had a strange issue. When I use [this product](https://www.amazon.com/MELIFE-Development-Dual-Mode-Microcontroller-Integrated/dp/B07Q576VWZ/ref=sr_1_1?dchild=1&keywords=esp32+melife&qid=1590114232&sr=8-1) The lines refresh one at a time rather than for each pixel. When I use the Sparkfun ESP32 Thing this doesn't happen. This probably isn't the fault of SmartMatrix but I don't really understand what would cause this. It's unfortunate because I would like to use generic boards. What's even stranger is that when I use the 64x64 matrix from Sparkfun with a generic ESP32 board I have no problems.

embedded-creations, 2020-05-22:
I don’t understand what you’re describing, some photos or a video would help. You probably need level shifters. Try lowering the i2s clock speed. Check the README for more details.

closed 2020-05-24 by Jkallus

embedded-creations, 2020-05-24:
We’re you able to fix the issue?  What worked?

Jkallus, 2020-05-24:
I wasn't able to fix this but suddenly the same issue started happening with the SparkFun ESP32 Thing. Here's a video of it https://drive.google.com/file/d/1DNgEazsZ4PNHrCylDeiUGfUpRZ2kbkHY/view?usp=sharing

It's supposed to be lighting each pixel one at a time but it seems to only refresh on the row change. The multiple text layers demo looks mostly alright other than the 16x32 panel being small for it. The other demo's don't seem to work. The 64x64 panel works fine for all of this so I'm just going to use that instead for the mean time. The 16x32 was from Adafruit so I would expect it to behave like most standard HUB75 panels. I didn't think there was anything actionable here so I thought to close the issue.