pixelmatix/SmartMatrix #123: Errors when using SmartMatrix teensylc in ESP-IDF

Jkallus, 2020-05-28:
Its unclear to me whether SmartMatrix for ESP32 is intended strictly for use within Arduino or with the ESP-IDF as well. I followed Espressif's instructions to reference the Arduino Core as an IDF component and I added SmartMatrix as a library with all the default Arduino ESP32 libraries. When I include SmartMatrix3.h and copy the MultipleTextLayers demo into app_main I get several errors. 

From Layer_Scrolling_Impl.h: 

Layer_Indexed_Impl.h:48:11: error: expected unqualified-id before '(' token
`     this->assert(indexedBitmap != NULL);`

From Layer_Scrolling_Impl.h:
Layer_Scrolling_Impl.h:43:11: error: expected unqualified-id before '(' token
     this->assert(scrollingBitmap != NULL);

From SmartMatrixMultiplexedCalcEsp32_Impl.h:
SmartMatrixMultiplexedCalcEsp32_Impl.h:489:15: error: 'void* memset(void*, int, size_t)' clearing an object of non-trivial type 'rgb48' {aka 'struct rgb48'}; use assignment or value-initialization instead [-Werror=class-memaccess]
         `memset(tempRow0, 0x00, sizeof(rgb48) * numPixelsPerTempRow);`

Same error for line 490, 809, and 809 in that file

Lastly there are several places which check if ESP32 is defined and intellisense appears to show that isn't defined so I am defining it myself at the top of my code. Not sure how proper that is. 

Sorry if this is the wrong place I just didn't know if anyone else has had this issue or whether this is supported. I was able to successfully use arduinoFFT in ESP-IDF so the ability to use arduino libraries in ESP-IDF in general appears to be supported.

embedded-creations, 2020-09-19:
Sorry it took months to respond, I'm not using ESP-IDF, but it seems like Espressif is ignoring the Arduino board support and focusing on IDF, so I'll probably get to it eventually.  LMK if you're still interested in ESP-IDF compatibility, and I'll try to look into these issues.

closed 2020-11-23 by embedded-creations

grene78, 2020-12-27:
I'm banging my head trying to get ESP-IDF & SmartMatrix to play nice, and I am not having much luck.  However, I am not well versed in ESP-IDF so it's likely user error.  Has anyone been able to use SmartMatrix 4 as an IDF component?