pixelmatix/SmartMatrix #126: Panel with chip  TC5020A doesn't work

Antoniohopu, 2020-08-28:
Hi, I'm trying to use your branch for esp32 support(https://github.com/rorosaurus/esp32-hub75-driver)  with a 16x32 panel led based on the TC5020A chips.
Due to  I'm not able to draw simple figures as for example a square or a circle, I would like to know if these libraries for esp32 and smart matrix panels are compatible with TC50250A chip and what are the steps that I should carry out for full compatibility between the libraries and esp32.

As an additional note, the libraries works well with adafruit 16x32 panels.
Thanks in advances.
Best Regards.
Antonio Rosa.

embedded-creations, 2020-08-28:
I'm not familiar with this chipset, but it's been reported elsewhere to require special addressing which SmartMatrix Library doesn't support:

https://github.com/hzeller/rpi-rgb-led-matrix/issues/1064

AndreasMainz, 2020-08-29:
I am using this chipset with a 64x64 panel and it works fine. I do not have special adressing..

Antoniohopu, 2020-08-31:
Hi, I have recently figured out that my 16x32 Chinese panel it's 1/4 scan. It is possible to use 1/4 scan configuration with your libraries and esp32?

embedded-creations, 2020-09-27:
Check out this thread in the forum: https://community.pixelmatix.com/t/smartmatrix-with-1-4-scan-hub75-panels/104/19