pixelmatix/SmartMatrix #131: HUB75 Controller for Multiple Platforms (Not just Teensy/ESP32)

embedded-creations, 2020-11-25:
I'm working on an open source solution that will allow a lot more platforms to refresh HUB75 panels.  If you want to know more, sign up for notifications on this GitHub Issue, or on [this thread](https://community.pixelmatix.com/t/standalone-matrix-controller-apa102-to-hub75-panel-adapter/532) in the SmartMatrix Community and I'll be sure to post more info when it's ready.

sukesh-g, 2021-02-24:
Hello Louis,

i have been waiting for smart matrix open source solution since more than one year. does it work for stm32.  may i know when will be release.

thanks
sukesh

embedded-creations, 2021-04-07:
@sukesh-g I have no plans to port SmartMatrix Library to STM32.  You could use the STM32 to send pixel data to an ESP32 which takes care of refreshing the HUB75 panel, that solution is described in the thread linked above

board707, 2022-02-08:
> i have been waiting for smart matrix open source solution since more than one year. does it work for stm32. may i know when will be release.

Hi sukesh
look at [this project](https://github.com/board707/DMD_STM32/tree/dev-V2)

FredWahl, 2022-10-12:
I hope it will be available for Raspberry pi.

embedded-creations, 2022-10-12:
@FredWahl Yes, it should support anything that can send APA102 data, including a RPi

juelketreu, 2024-01-16:
hi, i am trying the SmartMatrix for a GPS-clock. It work great! Thanks therefor. But i have a question to the HexColors: I will show the Outdoor-Temperture on the HUB75 in Colors similar to the Temp. What can i do to switch the colors from Hex to RGB in the Sketch like "setColor(25,155,255);" as it works in PxMatrix? Thanks a lot for your Help!

embedded-creations, 2024-01-16:
@juelketreu hi, this is a better question for the SmartMatrix Community, can you please ask there?

 https://community.pixelmatix.com/

juelketreu, 2024-01-16:
yes, of course :-)

greetings from Germany