pixelmatix/SmartMatrix #133: Monochrome 32*16 P10 HUB12 LED matrix control with ESP32

BodoMinea, 2020-12-01:
Hello,

From what I have read this library should be able to directly (no adapter/extra hardware) control a monochrome 32*16 P10 HUB12 LED matrix with an ESP32 microcontroller.

I have uploaded the following code to the ESP32-Devkit board:

```
#define SUPPORT_ADAFRUIT_GFX_LIBRARY

#include <MatrixHardware_ESP32_V0.h>
#include <SmartMatrix.h>

#define COLOR_DEPTH 24
const uint16_t kMatrixWidth = 32;
const uint16_t kMatrixHeight = 32;
const uint8_t kRefreshDepth = 36;
const uint8_t kDmaBufferRows = 4;
const uint8_t kPanelType = SMARTMATRIX_HUB12_16ROW_32COL_MOD4SCAN;
const uint32_t kMatrixOptions = (SMARTMATRIX_OPTIONS_HUB12_MODE);

SMARTMATRIX_ALLOCATE_BUFFERS(matrix, kMatrixWidth, kMatrixHeight, kRefreshDepth, kDmaBufferRows, kPanelType, kMatrixOptions);

#if 1
  const uint8_t kBackgroundLayerOptions = (SM_BACKGROUND_GFX_OPTIONS_NONE);
  SMARTMATRIX_ALLOCATE_BACKGROUND_GFX_LAYER(backgroundLayer, kMatrixWidth, kMatrixHeight, COLOR_DEPTH, kBackgroundLayerOptions);
#else
  const uint8_t kGFXMonoLayerOptions = (SM_GFX_MONO_OPTIONS_NONE);
  SMARTMATRIX_ALLOCATE_GFX_MONO_LAYER(backgroundLayer, kMatrixWidth, kMatrixHeight, kMatrixWidth, kMatrixHeight, COLOR_DEPTH, kGFXMonoLayerOptions);
#endif

#define BLACK   0x0000
#define RED     0xF800

void setup() {
  matrix.addLayer(&backgroundLayer); 
  matrix.begin();
  matrix.setBrightness(128);

  backgroundLayer.setTextSize(1);
  backgroundLayer.setTextColor(RED);
  backgroundLayer.println("Test");
  backgroundLayer.swapBuffers();
}

void loop() {
//  for(uint8_t rotation=0; rotation<4; rotation++) {
//    backgroundLayer.setRotation(rotation);
//    testText();
//    backgroundLayer.swapBuffers();
//    delay(1000);
//  }
}
```
In the MatrixHardware_ESP32_V0.h file I have selected the following option:
`#define GPIOPINOUT ESP32_FORUM_PINOUT_WITH_LATCH`

I made the following connections:

A - GPIO 5
B - GPIO 18
C - GPIO 19
CLK - GPIO 22
LAT - GPIO 26
R1 -   GPIO 2
OE -  GPIO 25

All GND's are grounded to the same ground as the power supply and devkit. E & D are not connected, as per the instructions at https://github.com/pixelmatix/SmartMatrix/wiki/ESP32-Port

On the back of the LED matrix panel, near the connector it is clearly written „HUB12” and the model of the matrix is „RK-P10(1R)-320*160-V2.0 V10”

My sketch is based on the Adafruit GFX example of the library and I am running on the 4.0.0 release from 5 days ago. I would also like to note that I encountered some compilation issues related to `error: 'textsize_y' was not declared in this scope smartmatrix`, in a graphic font related if-statement which I commented out (I do not have graphic fonts included).

Unfortunately the result shown on the matrix is this:
[image]
(the greenish dots in the lower side of the image are camera sensor reflections, only the white dots are actually on the screen)

Do you have any ideas of what fixes to try? Should I ask the vendor for the row scan pattern (1/2, 1/4 or 1/8) or does the library have different scan pattern options? (like the PxMatrix library has ZIGZAG, ZAGGIZ, WZAGZIG2, etc.)

Thank you very much in advance for your time looking into this and for developing such a good and useful library in the first place.

BodoMinea, 2020-12-01:
I would also like to note that the connections made are the same as @busel7 mentioned in #97 and also the code is similar. I guess something is different with my matrix given the fact that I can not display anything.

embedded-creations, 2020-12-01:
Do you mind reposting this to community.pixelmatix.com which is a better place to discuss troubleshooting issues?

BodoMinea, 2020-12-01:
Yeah, sure, no problem, I am reposting it right now. If you're part of the admin team of the aforementioned forum please note that external auth methods are broken:

Github: `Sorry, there was an error authorizing your account. Please try again.`

Google signup: **redirect_uri_mismatch** `The redirect URI in the request, https://community.pixelmatix.com/auth/google_oauth2/callback, does not match the ones authorized for the OAuth client. To update the authorized redirect URIs, visit: https://console.developers.google.com/apis/credentials/oauthclient/${your_client_id}?project=${your_project_number}`

BodoMinea, 2020-12-01:
As per your instructions, the thread was created here - https://community.pixelmatix.com/t/monochrome-32-16-p10-hub12-led-matrix-control-with-esp32/812

embedded-creations, 2020-12-01:
Thanks for the heads up, I just got Twitter back up and working, and Google now requires 4-6 weeks to verify my logo, ha

closed 2020-12-01 by embedded-creations