pixelmatix/SmartMatrix #134: How to ask for help

embedded-creations, 2020-12-01:
If you find a bug, or have a feature request, please post a GitHub Issue here

If you're looking for help with a project, or troubleshooting something, please post to [SmartMatrix Community](community.pixelmatix.com) as that's a better place to discuss these kind of topic, and leave it open for other people to find later

BodoMinea, 2020-12-27:
Just a little 'nudge' - the link for the community is wrong (it doesn't have the http/s protocol so it tries to find a GitHub issue named community.pixelmatrix.com).

Also, couldn't find a 'meta' section on the community for suggestions regarding the forum in itself so I'll point it out here. Could you enable the option for a dark theme in there? I see that the platform supports it if admins wish so - https://meta.discourse.org/t/allow-users-to-select-dark-light-theme/60857. Now that Github has dark theme, switching to the PixelMatrix community tab makes my (and probably others') eyes hurt.

Thanks in advance and keep up the good work!

embedded-creations, 2020-12-28:
@BodoMinea Thanks for the heads up on that broken link, now fixed.  I probably have done that elsewhere too...

I followed those instructions, see if the new Dark theme works.  I use a browser extension that applies dark themes to sites that don't support it, so it never bothered me