pixelmatix/SmartMatrix #138: Problem with the MuiltipleTextLayersGfx demo and FastLED

kazzle101, 2020-12-21:
Hello

I've been trying out the MultipleTextLayersGfx demo found in https://github.com/pixelmatix/SmartMatrix/blob/master/examples/MultipleTextLayersGfx/MultipleTextLayersGfx.ino

If I remove the comment for the line: #define INCLUDE_FASTLED_BACKGROUND

I get the following error:

`MultipleTextLayersGfx:50: error: expected constructor, destructor, or type conversion before '(' token
   SMARTMATRIX_ALLOCATE_BACKGROUND_GFX_LAYER(backgroundLayer, kMatrixWidth, kMatrixHeight, COLOR_DEPTH, kBackgroundLayerOptions);
                                            ^
In file included from C:\Users\karl\AppData\Local\Temp\arduino_modified_sketch_5334\MultipleTextLayersGfx.ino:138:0:
C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\FastLED/FastLED.h:14:21: note: #pragma message: FastLED version 3.003.003    #pragma message "FastLED version 3.003.003"
                     ^
MultipleTextLayersGfx: In function 'void setup()':
MultipleTextLayersGfx:187: error: 'backgroundLayer' was not declared in this scope
   matrix.addLayer(&backgroundLayer); 
                    ^
MultipleTextLayersGfx: In function 'void loop()':
MultipleTextLayersGfx:317: error: 'backgroundLayer' was not declared in this scope
   while(backgroundLayer.isSwapPending());
         ^

MultipleTextLayersGfx:319: error: 'backgroundLayer' was not declared in this scope
   rgb24 *buffer = backgroundLayer.backBuffer();
                   ^

expected constructor, destructor, or type conversion before '(' token`

I'm using the latest Arduino IDE/Teensdunio with SmartMatrix 4.0.3 installed.

Thanks
Karl

embedded-creations, 2020-12-21:
I just pushed a fix, hopefully you can use it as is and don't have to wait for the next release.  I checked that it compiles but didn't test as I don't have a test setup handy.  Please let me know if this works for you.

https://github.com/pixelmatix/SmartMatrix/commit/43ebc817417c4c404a5302dcc45b4e74766fb1f3

kazzle101, 2020-12-21:
Yes, that's working now, but I do see a couple of warning messages:

`MultipleTextLayersGfx: In function 'void loop()':
MultipleTextLayersGfx:339: warning: iteration 16u invokes undefined behavior 
       buffer[kMatrixWidth*j + i] = CRGB(CHSV(noise[j][i],255,noise[i][j]));
                                  ^
C:\Users\karl\AppData\Local\Temp\arduino_modified_sketch_795609\MultipleTextLayersGfx.ino:334:20: note: containing loop
   for(int i = 0; i < kMatrixWidth; i++) {
`

Thanks
Karl.

embedded-creations, 2020-12-21:
Glad it’s working. That warning is related to #81, I think the math (which I didn’t write and don’t understand at a glance) is limited to a smaller width than is commonly used with SmartMatrix Library now.

closed 2020-12-21 by embedded-creations