pixelmatix/SmartMatrix #140: no matching function for call to 'nblend(rgb24&, CRGB&, int)'

alpha815, 2021-01-01:
i was trying to port fastled pride2015 example for smartmatrix following this guideline
[](https://github.com/pixelmatix/SmartMatrix/blob/master/MIGRATION.md#porting-fastled-sketch-to-smartmatrix)
after making some changes in code i tried to complile sketch and encontered following error.
here is code 
`#include <FastLED.h>
#include <SmartMatrix3.h>
// Pride2015
// Animated, ever-changing rainbows.
// by Mark Kriegsman
//smartmatrix3
#define COLOR_DEPTH 24                  // This sketch and FastLED uses type `rgb24` directly, COLOR_DEPTH must be 24
const uint8_t kMatrixWidth = 32;        // known working: 32, 64, 96, 128
const uint8_t kMatrixHeight = 16;       // known working: 16, 32, 48, 64
const uint8_t kRefreshDepth = 24;       // known working: 24, 36, 48
const uint8_t kDmaBufferRows = 2;       // known working: 2-4, use 2 to save memory, more to keep from dropping frames and automatically lowering refresh rate
const uint8_t kPanelType = SMARTMATRIX_HUB75_16ROW_MOD8SCAN;   // use SMARTMATRIX_HUB75_16ROW_MOD8SCAN for common 16x32 panels
const uint8_t kMatrixOptions = (SMARTMATRIX_OPTIONS_NONE);      // see http://docs.pixelmatix.com/SmartMatrix for options
//const uint8_t kBackgroundLayerOptions = (SM_BACKGROUND_OPTIONS_NONE);
const uint8_t kBackgroundLayerOptions = (SM_BACKGROUND_OPTIONS_NONE);
const uint8_t kScrollingLayerOptions = (SM_SCROLLING_OPTIONS_NONE);

SMARTMATRIX_ALLOCATE_BUFFERS(matrix, kMatrixWidth, kMatrixHeight, kRefreshDepth, kDmaBufferRows, kPanelType, kMatrixOptions);
SMARTMATRIX_ALLOCATE_BACKGROUND_LAYER(backgroundLayer, kMatrixWidth, kMatrixHeight, COLOR_DEPTH, kBackgroundLayerOptions);
SMARTMATRIX_ALLOCATE_SCROLLING_LAYER(scrollingLayer, kMatrixWidth, kMatrixHeight, COLOR_DEPTH, kScrollingLayerOptions);

#define BRIGHTNESS  255
#define NUM_LEDS    512
CRGB leds[NUM_LEDS];

void setup() {
  delay(3000); // 3 second delay for recovery
  
  matrix.addLayer(&backgroundLayer);
  // matrix.addLayer(&scrollingLayer);
  matrix.begin();
  matrix.setBrightness(BRIGHTNESS);
}

void loop()
{
  pride();
  backgroundLayer.swapBuffers();
}

// This function draws rainbows with an ever-changing,
// widely-varying set of parameters.
void pride() 
{
  rgb24 *buffer = backgroundLayer.backBuffer();
  static uint16_t sPseudotime = 0;
  static uint16_t sLastMillis = 0;
  static uint16_t sHue16 = 0;
 
  uint8_t sat8 = beatsin88( 87, 220, 250);
  uint8_t brightdepth = beatsin88( 341, 96, 224);
  uint16_t brightnessthetainc16 = beatsin88( 203, (25 * 256), (40 * 256));
  uint8_t msmultiplier = beatsin88(147, 23, 60);

  uint16_t hue16 = sHue16;//gHue * 256;
  uint16_t hueinc16 = beatsin88(113, 1, 3000);
  
  uint16_t ms = millis();
  uint16_t deltams = ms - sLastMillis ;
  sLastMillis  = ms;
  sPseudotime += deltams * msmultiplier;
  sHue16 += deltams * beatsin88( 400, 5,9);
  uint16_t brightnesstheta16 = sPseudotime;
  
  for( uint16_t i = 0 ; i < NUM_LEDS; i++) {
    hue16 += hueinc16;
    uint8_t hue8 = hue16 / 256;

    brightnesstheta16  += brightnessthetainc16;
    uint16_t b16 = sin16( brightnesstheta16  ) + 32768;

    uint16_t bri16 = (uint32_t)((uint32_t)b16 * (uint32_t)b16) / 65536;
    uint8_t bri8 = (uint32_t)(((uint32_t)bri16) * brightdepth) / 65536;
    bri8 += (255 - brightdepth);
    
    CRGB newcolor = CRGB(CHSV( hue8, sat8, bri8));
    
    uint16_t pixelnumber = i;
    pixelnumber = (NUM_LEDS-1) - pixelnumber;
    
    nblend( buffer[pixelnumber], newcolor, 64);
  }
}`

embedded-creations, 2021-01-01:
What version of SmartMatrix Library are you running?  You're including SmartMatrix3.h so it seems out of date.  I'm reluctant to help you with an out of date version as the fix may be different for the latest released version (4.0.3).  I'm also reluctant to ask you to update as it may break other code that you currently have working.

Maybe you can make a copy of the "SmartMatrix3" folder in Arduino/libraries so you can copy it back and go back to the version that's working with your other code if needed?  You should be able to download the latest release and install it into libraries/SmartMatrix (not SmartMatrix3) and have both working in parallel.  You'll need to update your sketches to use SmartMatrix.h and maybe other minor changes, see [MIGRATION.md](https://github.com/pixelmatix/SmartMatrix/blob/master/MIGRATION.md).  You may also need to update Marc's SmartMatrix::GFX library.

embedded-creations, 2021-01-01:
Anyway, I found it's an easy fix, include FastLED after SmartMatrix.  That's it

closed 2021-01-01 by embedded-creations

alpha815, 2021-01-02:
it is working now thanks