pixelmatix/SmartMatrix #143: ESP32  GifDecoder Issue

InnoGreenTech, 2021-01-24:
Hello, Thank you for your work.

I try to use your GifDecoder with an ESP 32 Wroom and a matrix LED 64x32.  In loop i have this issue. Can you help me?

Best regards

21:58:24.624 -> Rebooting...
21:58:24.671 -> ets Jun  8 2016 00:22:57
21:58:24.671 -> 
21:58:24.671 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
21:58:24.671 -> configsip: 0, SPIWP:0xee
21:58:24.671 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
21:58:24.671 -> mode:DIO, clock div:1
21:58:24.671 -> load:0x3fff0018,len:4
21:58:24.671 -> load:0x3fff001c,len:1216
21:58:24.671 -> ho 0 tail 12 room 4
21:58:24.671 -> load:0x40078000,len:9720
21:58:24.671 -> ho 0 tail 12 room 4
21:58:24.671 -> load:0x40080400,len:6352
21:58:24.671 -> entry 0x400806b8
21:58:25.888 -> Starting AnimatedGIFs Sketch
21:58:25.888 -> 

21:58:25.888 -> Starting SmartMatrix Mallocs

21:58:25.888 -> Heap/32-bit Memory Available: 326472 bytes total, 113792 bytes largest free block
21:58:25.888 -> 8-bit/DMA Memory Available  : 262960 bytes total, 113792 bytes largest free block
21:58:25.888 -> Total PSRAM used: 0 bytes total, 0 PSRAM bytes free
21:58:25.888 -> assertion "backgroundBuffers[0] != NULL" failed: file "C:\Users\fabri\OneDrive\Arduino\Arduino\libraries\SmartMatrix\src/Layer_BackgroundGfx_Impl.h", line 58, function: void SMLayerBackgroundGFX<RGB, optionFlags>::begin() [with RGB = rgb24; unsigned int optionFlags = 0u]
21:58:25.888 -> abort() was called at PC 0x400e73fb on core 1
21:58:25.888 -> 
21:58:25.888 -> Backtrace: 0x4008dc8c:0x3ffb1ec0 0x4008deb9:0x3ffb1ee0 0x400e73fb:0x3ffb1f00 0x400d16fa:0x3ffb1f30 0x400d1f3b:0x3ffb1f50 0x400d236c:0x3ffb1f80 0x400d6b4f:0x3ffb1fb0 0x4008a649:0x3ffb1fd0
21:58:25.888 ->

embedded-creations, 2021-01-24:
Looks like there’s not enough memory to set up the SmartMatrix buffers. Are you using the AnimatedGIFs example, or some other code?

this is a better discussion for the forum, please see the pinned GitHub Issue

InnoGreenTech, 2021-01-25:
Thank you yes i have used the AnimatesGIFs example, I am setting the pinned with ESP32_JC_RIBBON_PINOUT_WEMOS 
Sorry I do not find the discussion on  forum.