pixelmatix/SmartMatrix #146: SmartMatrix-4.0.3   #pragma GCC error

C4sperT3ch, 2021-04-27:
Need Help

Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, Disabled, Default 4MB with spiffs (1.2MB APP/1.5MB SPIFFS), 240MHz (WiFi/BT), QIO, 80MHz, 4MB (32Mb), 921600, None"

In file included from C:\Users\SERVER TEI\Documents\Arduino\libraries\SmartMatrix-4.0.3\examples\MultipleTextLayers\MultipleTextLayers.ino:25:0:

C:\Users\SERVER TEI\Documents\Arduino\libraries\SmartMatrix-4.0.3\src/SmartMatrix.h:74:19: error: No MatrixHardware*.h file included - You must include one at the top of your sketch

 #pragma GCC error "No MatrixHardware*.h file included - You must include one at the top of your sketch"

                   ^

embedded-creations, 2021-04-27:
The error gives you a hint: `"No MatrixHardware*.h file included - You must include one at the top of your sketch"`

More details: https://github.com/pixelmatix/SmartMatrix/wiki/Features#init

closed 2021-04-27 by embedded-creations

C4sperT3ch, 2021-04-28:
How to